OctoFarm/OctoFarm #8: File Manager Crashes when trying to open when no printers connected.

TheKyleJohnson90, 2020-03-15:
It seems the File Manager tab will crash if there are not active connected printers. The following error was shown.
```

TypeError: /home/ubuntu/OctoFarm/views/filemanager.ejs:48
    46|         </center>
    47|         <div id="printerList" class="list-group">
 >> 48|           <%- include('./partials/FileManager/printerList.ejs') %>
    49|         </div>
    50|       </div>
    51|       <div class="card-footer fileFooter text-muted"></div>

/home/ubuntu/OctoFarm/views/partials/FileManager/printerList.ejs:66
    64|         </div>
    65|         <small class="pt-2 float-left"
 >> 66|           ><i class="fas fa-cube"></i> <b>H:</b> <%=
    67|           printer.profiles[selectedProfile].volume.height%>mm x <b>W:</b> <%=
    68|           printer.profiles[selectedProfile].volume.width%>mm x <b>D:</b> <%=
    69|           printer.profiles[selectedProfile].volume.depth%>mm</small

Cannot read property '_default' of undefined
    at printers.forEach (/home/ubuntu/OctoFarm/views/partials/FileManager/printerList.ejs:65:27)
    at Array.forEach (<anonymous>)
    at eval (/home/ubuntu/OctoFarm/views/partials/FileManager/printerList.ejs:10:17)
    at printerList (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:656:17)
    at include (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:654:39)
    at eval (/home/ubuntu/OctoFarm/views/filemanager.ejs:13:17)
    at filemanager (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:656:17)
    at tryHandleCache (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:254:36)
    at View.exports.renderFile [as engine] (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:459:10)
    at View.render (/home/ubuntu/OctoFarm/node_modules/express/lib/view.js:135:8)
    at tryRender (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:640:10)
    at Function.render (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/home/ubuntu/OctoFarm/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/home/ubuntu/OctoFarm/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at router.get (/home/ubuntu/OctoFarm/routes/index.js:59:7)
    at <anonymous>
```

NotExpectedYet, 2020-03-15:
That looks more like it's trying to load a printer that has never been on the farm as the initial scan should collect that metric. Any printer not offline should have already grabbed it, and file manager won't load offline printers. 

Could you give me more information about the state of the printers? like are they all offline and never been connected, some offline and some connected?

TheKyleJohnson90, 2020-03-16:
This was right after a fresh start of OctoFarm; I had just finished adding new printers. I had not connected any of the printers yet and tried to go into the File Manager

NotExpectedYet, 2020-03-16:
That makes sense. Cheers.

commit 18a5b069 (2020-03-18): Fixed File Manager ejs error

Fixes: #8
closed 2020-03-29 by NotExpectedYet