OctoFarm/OctoFarm #11: Printer "stuck"

noxin1, 2020-03-18:
Reviewing the container logs I get the log snippet below.   The printer did get halted by the firmware due to a heating issue.  The printer was restarted and has been printing happily since.

-- Log --
(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.

(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.



e

(node:35) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'tool0' of null

    at Function.complete (/app/runners/history.js:28:40)

    at runMicrotasks (<anonymous>)

    at processTicksAndRejections (internal/process/task_queues.js:93:5)

(node:35) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)

(node:35) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

PrintFailed



e

(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.

(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.

(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.



e

(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.

(node:35) TimeoutOverflowWarning: 1.7976931348623157e+308 does not fit into a 32-bit signed integer.

Timer duration was truncated to 2147483647.

noxin1, 2020-03-18:
Restarted the docker container, issue remains.

NotExpectedYet, 2020-03-18:
I've been chasing that one down myself because there's nothing in the printFailed function that would request tool0. The truncate issues is a nothing burger and can be ignored. 

It looks like it fired off a print success then a print failed for some strange reason. Restarting the docker container should just continue on as normal though. 

I'll do some more tests and see if I can figure out what's going on. 

Just to clarify the print failed as a Error and not a cancel?

NotExpectedYet, 2020-03-29:
Let me know if you still experience this on latest man, I personally think (hope) it's sovled ahah

NotExpectedYet, 2020-04-05:
Closing as this should now finally be resolved in the latest release.

closed 2020-04-05 by NotExpectedYet