OctoFarm/OctoFarm #15: _default profile, and circular reference issue

NotExpectedYet, 2020-03-29:
I've found the circular reference issue, 

This doesn't stop functioning but needs looking into and I believe it's related to the _default profile issue I've been seeing.

NotExpectedYet, 2020-03-29:
Narrowing it down, both these are specific to Offline printers only. Offline, never added to the farm by the looks of it.

Crowlord, 2020-03-30:
The UI directs me towards this issue but I am not sure which part of the profile you need. 

The profile for my Mk.5S and Mk3S are not found. 

Files on all 3 printers are not found.

NotExpectedYet, 2020-03-30:
What are you running OctoPrint on? 

Also what OctoPrint version please. 

I'm still trying to chase down this issue as I can't reproduce. 

If you can backup your config I can put it on my test instance and try as there's only been limited people with this issue so far.

Crowlord, 2020-03-30:
3 Instances 

3B 1.2 running Octoprint 1.4.0 on octopi 0.15.1- printer is Prusa Mk2.5S
Reports cannot scan profile unable to change any settings
3B+ 1.3 running Octoprint 1.4.0 on Octopi 0.16.0 - printer is  Prusa Mk3S
Reports cannot scan profile unable to change any settings
3B+ 1.3 running Octoprint 1.4.0 on octopi 0.16.0 - Printer Anycubic Kossel Linear Plus
Scans profile and can change settings will not list files. 

Running all component for Farm on Win 10 machine. 

Prusas do have custom bounding boxes set to account for out of print bed priming line. 

Not sure which config you need?

NotExpectedYet, 2020-03-31:
I doubt it's the custom bounding box. 

There's a plugin that lets you backup your settings. If you grab me that I can double check against your specific config. 

Just to check, CORS is enabled on all these printers, and you've made sure OctoFarm connected WITH CORS correct?

NotExpectedYet, 2020-03-31:
Also, files do not work with SD Card. Only local storage just to add that.

Crowlord, 2020-03-31:
OK. CORS is enabled and to the best of my knowledge connecting.
[octoprint-backup-20200331-072344.zip](https://github.com/NotExpectedYet/OctoFarm/files/4407133/octoprint-backup-20200331-072344.zip)
[octoprint-backup-20200331-072236.zip](https://github.com/NotExpectedYet/OctoFarm/files/4407134/octoprint-backup-20200331-072236.zip)
[octoprint-backup-20200331-072423.zip](https://github.com/NotExpectedYet/OctoFarm/files/4407135/octoprint-backup-20200331-072423.zip)

NotExpectedYet, 2020-03-31:
Thanks for those files, I won't get chance to replicate until mid way through the week but I'll get back to you hopefully with a fix when I do. 

Your files should still scan through though if CORS is enabled, would you mind pasting the production.log and productionError.log's into pastebin and sending the link here please?

I'll double check that isn't related to the profile issue.

Crowlord, 2020-03-31:
https://pastebin.com/rpssPWZD

production error is just ^C

NotExpectedYet, 2020-03-31:
No errors at all then, and according to the logs it successfully grabbed your files so it must be the profile issue. Will get back to you when I've tested that further. I thought it might be AUTO setting but won't reporduce on my setup either. -_-

NotExpectedYet, 2020-03-31:
Can someone please screenshot the browser console output after they've opened Printer Manager and got it not loading. It might allude me to where this issue originates from.

NotExpectedYet, 2020-03-31:
Also, could you either of you try booting in Safe mode too? 

And get me a list of plugins you have, I'm wondering if it's down to that due to fresh installs working from other users reports.

nkatsoulis, 2020-03-31:
Not sure if this helps but It was working on the last 1.1.3 version as far as I recall.

NotExpectedYet, 2020-03-31:
Sadly it still persists as I'd only had PiZero reports of this. Have you tried safe mode? I'm wondering if it's maybe a plugin and that's why I can't reproduce.

nkatsoulis, 2020-03-31:
> Sadly it still persists as I'd only had PiZero reports of this. Have you tried safe mode? I'm wondering if it's maybe a plugin and that's why I can't reproduce.

Just did this and no change.

nkatsoulis, 2020-03-31:
Are there supposed to be logs in the logs folder? i have nothing on my persistent mount.  Im connected to my container and not seeing much for logs.

nkatsoulis, 2020-03-31:
FYI im on discord as GIMP if you want to message me there when you have some time.  I can get you access to assist in troubleshooting

NotExpectedYet, 2020-03-31:
Not on docker, I've not properly set it up yet which I'm gathering your using? 

I couldn't get my command to run the npm start script but I'll get that fixed. 

The logs won't show much as it only registers a fail on connection, not that it couldn't find the file and from other users reports it's pulling the information properly. 

I'm at a loss currently to why some users experience this and others don't but I'll get more time through the week to have a look.

NotExpectedYet, 2020-03-31:
@nkatsoulis thanks for the offer too, if you can pass me setting configs I can replicate on my 60 testing instances and take a much deeper look at it. I've got a PiZero I'm going to get spun up this week too. if I sort this I'll do a bug fix release to finalise it all as well.

enderjones, 2020-03-31:
I am having a similar issue.  It will not show any information under my printer profiles.  And under the file manager i show no files on any printer with this message "No profile found matching your Profile".  I am running all my printers on separate pizeros.  Is this a PiZero issue?

NotExpectedYet, 2020-04-01:
@enderjones  I thought it was limited to PiZero's at first but it seems other users are having the same issue. I'm hoping to have a fix sometime this week for it though. 

I re-added the throttle back into the latest master too, and another user got his PiZero's working with a fresh install. His seems to be because of a plugin but I'm not sure which. I'll be sure to update here with my testing progress through the week when I've setup my PiZero I've got here. It's a bit of an annoying one to track down at the minute there doesn't seem to be much rhyme or reason to it. I'll figure it out though, it's more than likely related to that circular reference that keeps cropping up but there's a lot of info to dig through.

NotExpectedYet, 2020-04-01:
I'll support PiZero if possible and release recommendations with it once I've made sure myself it's all fine and I've ironed out those bugs.

NotExpectedYet, 2020-04-01:
Hey everyone, 

I finally reproduced this issue! Was just setting up my new IDE and I will have a fixed pushed for the next release. Thanks for all the comments they helped me narrow it down. It's not plugins, and I haven't tested yet but it isn't the PiZero either. 

Will be coming in a bug fix release before the end of this week!

enderjones, 2020-04-03:
@NotExpectedYet Thanks for the update and being diligent on this project.  I think I can speak for everyone, we are extremely excited for this to work properly!

JamesVexica, 2020-04-03:
@enderjones no trouble at all :) It's becoming my full time job after march because I'm been furloughed from work here in the UK due to corona. I will be pushing a more consistent release schedule then too. Stability will be my main focus in the coming weeks, and updating a lot of the flow. Hopefully some documentation too aha.

JamesVexica, 2020-04-03:
Woops, still logged into my work account ahah

NotExpectedYet, 2020-04-03:
@enderjones no trouble at all :) It's becoming my full time job after march because I'm been furloughed from work here in the UK due to corona. I will be pushing a more consistent release schedule then too. Stability will be my main focus in the coming weeks, and updating a lot of the flow. Hopefully some documentation too aha.

closed 2020-04-04 by NotExpectedYet

enderjones, 2020-04-04:
@NotExpectedYet Thank you!

NotExpectedYet, 2020-04-04:
> @NotExpectedYet Thank you!

No worries! I have tested and re-tested man this should have completely squashed that issue!

commit f901363d (2020-04-04): Fixed issue with not collecting profiles under certain conditions...

Fixed #15
commit 319de2fd (2020-04-04): Fixed issue with not collecting profiles under certain conditions...

Fixed #15
commit fcb4c9b8 (2020-04-04): Fixed issue with not collecting profiles under certain conditions...

Fixed #15