OctoFarm/OctoFarm #16: Cannot reach OctoFarm using Docker

mlagerweij, 2020-03-30:
Hello,

Portainer shows the container/image to be up and running, but for whatever reason the OctoFarm instance can't be reached on <ip>:4000

Container was created as per instructions on these pages. 

> `This site can’t be reached192.168.1.200 refused to connect.`

Any suggestions?

Thanks!

NotExpectedYet, 2020-03-30:
There's a few instructions for docker on here could you be more specific please?

mlagerweij, 2020-03-30:
Using the line:

docker run -d --name octofarm -e "MONGO=mongodb://172.17.0.2/octofarm" -p4000:4000 octofarm/octofarm -v '<your persistent folder>/OctoFarm/config/':'/app/serverConfig/':'rw' -v '<your persistent folder>/OctoFarm/logs/':'/app/logs/':'rw'

with the `<your persistent folder>` changed to `/home/mark `

mlagerweij, 2020-03-30:
Using the docker compose options gives the same result.

NotExpectedYet, 2020-03-30:
Have you setup mongo seperately? That looks like an internal docker IP although I've  how portainer routes those. 

Can you get onto the docker console window as well you'll see what's happening in there as it will output some logs

mlagerweij, 2020-03-30:
Something like this?:

 ejs@3.0.1 postinstall /app/node_modules/ejs
> node ./postinstall.js

Thank you for installing EJS: built with the Jake JavaScript build tool (https://jakejs.com/)

> nodemon@2.0.2 postinstall /app/node_modules/nodemon
> node bin/postinstall || exit 0

Love nodemon? You can now support the project via the open collective:
 > https://opencollective.com/nodemon/donate

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.1.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.1.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"x64"})

added 276 packages from 220 contributors and audited 476 packages in 4.722s
found 0 vulnerabilities

/app
Config already exists
db: mongodb://172.20.0.3/octofarm

mlagerweij, 2020-03-30:
octofarm-docker_mongodb_1 | running |   | octofarm-docker | mongo:latest | 2020-03-30 13:54:14 | 172.20.0.3 | 27017:27017

NotExpectedYet, 2020-03-30:
That's not able to connect to your mongodb. How have you set up mongodb?

NotExpectedYet, 2020-03-30:
I don't think docker routes the internal ip but I've no idea as I've not much experience with it.

NotExpectedYet, 2020-03-30:
doesn't it use hostnames not IP addresses?

mlagerweij, 2020-03-30:
I'm not an expert either. So far I have been able to connect to the OctoPrint instances and camera's using the IP addresses.

mlagerweij, 2020-03-30:
Mongo was installed using docker-compose:

  mongodb:
    image: mongo:latest
    environment:
      MONGO_INITDB_ROOT_USERNAME: root
      MONGO_INITDB_ROOT_PASSWORD: rootpassword
      MONGO_INITDB_DATABASE: octofarm
    ports:
      - 27017:27017
    volumes:
      - mongodb-data:/data/db
    restart: always

NotExpectedYet, 2020-03-30:
You have put authentication on your mongodb instance so that's probably why. Your URL string will need updating with those credentials I think. If you look at the example in my file for an external one you should be able to just copy the way I set the username and password up. Apparently docker been on the same host should route the IP fine but it risks it changing on reboot.

mlagerweij, 2020-03-30:
That does the trick...thank you!

NotExpectedYet, 2020-03-31:
Sweet closing :) I'll update the documentation too.

closed 2020-03-31 by NotExpectedYet