OctoFarm/OctoFarm #17: 1.1.4 - Unable to look at printer preferences.

nkatsoulis, 2020-03-31:
Printer is on pi version 1.3.12

[image]

I can't seem to get any details in the control or other screens.

[image]

I have used the latest with the following compose file below.

version: "2"
services:
  octofarm:
    container_name: octofarm
    image: octofarm/octofarm
    restart: always
    ports:
      - 4000:4000
    environment:
      - MONGO=mongodb://octofarm_mongo/octofarm
    volumes:
      - /volume1/homes/admin/data/octofarm/config:/app/serverConfig
      - /volume1/homes/admin/data/octofarm/logs:/app/logs

  octofarm_mongo:
    container_name: octofarm_mongo
    image: bitnami/mongodb:latest
    restart: always
    volumes:
      - /volume1/homes/admin/data/bitnami:/bitnami
    environment:
      - ALLOW_EMPTY_PASSWORD=yes
      - MONGODB_EXTRA_FLAGS=--wiredTigerCacheSizeGB=2

NotExpectedYet, 2020-03-31:
I believe this issue is related to the profile issue I'm seeing with some users. 

Issue #15 

I can't reproduce personally and I don't know what's causing the issue yet. 

Will report back once I've had to test further.

NotExpectedYet, 2020-03-31:
Closing to track further on #15

closed 2020-03-31 by NotExpectedYet