OctoFarm/OctoFarm #20: Not able to Save printer settings, cannot remotely start print job either.

jbishop129, 2020-04-01:
Hi,
First.. great concept! I recently added a second printer, and happened upon your project - exactly what I needed. I'm able to add printers in via IP, port + API key, and it auto-configs the camera stream. Great! But after that, I cannot go in and make any changes to the printer in OctoFarm. For example, giving the printer a name, or setting the webcam rotation checkboxes, when you hit Save, it just "spins" forever. 

Secondly, I cannot seem to initiate a print thru the interface. Clicking Print seems to do nothing. I tried using the API key found in the OctoPrint settings panel, I tried generating a custom app API key as admin, same issue. Meanwhile I can initiate a remote print via Cura, so it seems the issue is just with OctoFarm.

Any ideas on either of these issues?

Running on Docker, deployed via Swarm. Verified it is able to r/w the volume mounts and so on.

NotExpectedYet, 2020-04-01:
Thanks! 

Have you enabled CORS and restarted OctoPrint? 

Also is the client computer you are accessing it from able to communicate with the lan your OctoPrint instances are on? the web browser your accessing it from makes those connections and I only mention it because I've no idea how docker swarm works.

NotExpectedYet, 2020-04-05:
Closing this because I think your issue is how your accessing OctoFarm. It's not setup for remote access of control at the moment but I do hope to support this in a future release. Please re-open if not the case.

closed 2020-04-05 by NotExpectedYet