OctoFarm/OctoFarm #22: Filemanager crashes when using Canvas Hub

TheyCallMeZ, 2020-04-05:
I have a Canvas Hub that I use sometimes without the Palette 2 pro, when I have it in my printer list on octofarm (does not matter if it is by itself or with other printers) I get the following error when clicking file manager

`TypeError: /app/views/filemanager.ejs:48
    46|         </center>
    47|         <div id="printerList" class="list-group">
 >> 48|           <%- include('./partials/FileManager/printerList.ejs') %>
    49|         </div>
    50|       </div>
    51|       <div class="card-footer fileFooter text-muted"></div>

/app/views/partials/FileManager/printerList.ejs:72
    70|         <% if(selectedProfile != null && typeof printers[sorted.actualIndex].profiles != 'undefined'){ %>
    71|           <small class="pt-2 float-left"
 >> 72|           ><i class="fas fa-cube"></i> <b>H:</b> <%=
    73|           printers[sorted.actualIndex].profiles[selectedProfile].volume.height%>mm x <b>W:</b> <%=
    74|           printers[sorted.actualIndex].profiles[selectedProfile].volume.width%>mm x <b>D:</b> <%=
    75|           printers[sorted.actualIndex].profiles[selectedProfile].volume.depth%>mm</small

Cannot read property 'volume' of undefined
    at eval (/app/views/partials/FileManager/printerList.ejs:75:66)
    at Array.forEach (<anonymous>)
    at eval (/app/views/partials/FileManager/printerList.ejs:8:20)
    at printerList (/app/node_modules/ejs/lib/ejs.js:662:17)
    at include (/app/node_modules/ejs/lib/ejs.js:660:39)
    at eval (/app/views/filemanager.ejs:11:17)
    at filemanager (/app/node_modules/ejs/lib/ejs.js:662:17)
    at tryHandleCache (/app/node_modules/ejs/lib/ejs.js:255:36)
    at View.exports.renderFile [as engine] (/app/node_modules/ejs/lib/ejs.js:461:10)
    at View.render (/app/node_modules/express/lib/view.js:135:8)
    at tryRender (/app/node_modules/express/lib/application.js:640:10)
    at Function.render (/app/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/app/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/app/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at /app/routes/index.js:85:7
    at runMicrotasks (<anonymous>)`

NotExpectedYet, 2020-04-05:
Does the hub make any modifications to your profile in OctoPrint? 

I don't have one, so I can't check this myself.

TheyCallMeZ, 2020-04-05:
I don't have the direct answer for you unfortunately, however I can tell you that you can DIY a hub (which is what I did) 
the article for it is here

https://support.mosaicmfg.com/hc/en-us/articles/360012423734-Setup-Guide-DIY-CANVAS-Hub-OctoPi-CANVAS-and-P2-Plugins-

NotExpectedYet, 2020-04-05:
The plugin works fine, I have just checked with another user. What version of OctoFarm and what version of OctoPrint please? 

If it's specifically when the hub is plugged into your Pi, I've no way to test and no way to resolve sorry so let's hope it's something else. 

Could you also show me this screen in Printer Manager?: 
[image]
If file manager breaks, so would this at that specific line.

TheyCallMeZ, 2020-04-05:
my apologies, I will have to get this to you later tonight.

NotExpectedYet, 2020-04-05:
no worries, just let me know and I'll see if I can get to the bottom of it. 

the re-sync button will make sure all that's in place it might be worth running incase it's missing something.

TheyCallMeZ, 2020-04-05:
so if I understand what you want, it is from the Dashboard and clicking the
gear by the one in question
[image: image.png]

TheyCallMeZ, 2020-04-06:
One more thing... So I dug up a Pi that I had that I used to use on my
CR-10 S5 which had a vanilla instance of OctoPrint.
it comes up just fine despite the fact that the printer profile was reset
somehow..
[image: image.png]

But, when I go to click on the other one which right now is the PrusaPrint
image of OctoPrint running on the Pi Zero W, it still brings up the one for
the CR-10 S5. So whatever is affecting the Canvas Hub version might also be
an issue with the PrusaPrint version. Please let me know what I can do to
help.

TheyCallMeZ, 2020-04-06:
Sorry for the additional reply, on a whim I clicked file manager and it
opened... so yeah something doesn't seem to be parsing the other instances
correctly or something to that effect.

NotExpectedYet, 2020-04-06:
PrusaPrint image of OctoPrint? 
Yeah I'm gonna probably say that isn't supported. I've not got one and no idea what OctoPrint changes they've made. Be impossible without having one and supporting anything but OctoPrint isn't on the roadmap yet. 
I can't see those images you attached? But yeah the dashboard cog Printer Page please. I gather it's going to say something like undefined. If the Canvas hub is modifying that I've no way currently to test how it changes it, I'd also have to have some kind of indication it's connected to the device. 
Are you on the latest bug fix to confirm? You will see 1.1.4.2 in the System dropdown next to settings. I presumed it would be my system not picking up the profile correctly as I've had issues with that in the past which as far as I'm aware have been resolved for anyone having the issue.

NotExpectedYet, 2020-04-06:
For the next version I'll add in some checks to make the file manager not so reliant on that information, which I thought I'd already done but obviously this change must happen at a deeper location on that object. I can do it by the logs originally provided and hopefully that get's it up and running. You'll just not be able to see bed size and that in the file manager.

TheyCallMeZ, 2020-04-06:
Image for the OctoPrint with the canvas hub plugins
[image]

My Vanilla Octoprint
[image]

Running Version 1.1.42
[image]

NotExpectedYet, 2020-04-06:
Sweet cheers, yeah I'll have to see if those tweaks I make to allow for that information to be ignored will help. I'll add a push note for this issue for you when it's been pushed to development you are welcome to test it when it's done.

TheyCallMeZ, 2020-04-06:
Cool! I look forward to it.

NotExpectedYet, 2020-04-06:
I'm sure it will sort it :) You'll just see a message in there stating details cannot be found or something similar. As for actually supporting the thing to present that information will be a little tougher without one aha.

NotExpectedYet, 2020-04-06:
@TheyCallMeZ  

When you get a Mo, I've just pushed a potential fix to development. The docker will build and update shortly after if that's what your using. :) 

Let me know if it works.

TheyCallMeZ, 2020-04-06:
Version 1.1.4.3 (Cloned the repo and switched to development branch) I had to remove the machines that were non-vanilla to be able to get the file manager to open. 

TypeError: c:\git\OctoFarm\views\filemanager.ejs:48
    46|         </center>

    47|         <div id="printerList" class="list-group">

 >> 48|           <%- include('./partials/FileManager/printerList.ejs') %>

    49|         </div>

    50|       </div>

    51|       <div class="card-footer fileFooter text-muted"></div>

c:\git\OctoFarm\views\partials\FileManager\printerList.ejs:72
    70|         <% if(selectedProfile != null && typeof printers[sorted.actualIndex].profiles != 'undefined'){ %>

    71|           <small class="pt-2 float-left"

 >> 72|           ><i class="fas fa-cube"></i> <b>H:</b> <%=

    73|           printers[sorted.actualIndex].profiles[selectedProfile].volume.height%>mm x <b>W:</b> <%=

    74|           printers[sorted.actualIndex].profiles[selectedProfile].volume.width%>mm x <b>D:</b> <%=

    75|           printers[sorted.actualIndex].profiles[selectedProfile].volume.depth%>mm</small

Cannot read property 'volume' of undefined
    at eval (c:\git\OctoFarm\views\partials\FileManager\printerList.ejs:75:66)
    at Array.forEach (<anonymous>)
    at eval (c:\git\OctoFarm\views\partials\FileManager\printerList.ejs:8:20)
    at printerList (c:\git\OctoFarm\node_modules\ejs\lib\ejs.js:662:17)
    at include (c:\git\OctoFarm\node_modules\ejs\lib\ejs.js:660:39)
    at eval (c:\git\OctoFarm\views\filemanager.ejs:11:17)
    at filemanager (c:\git\OctoFarm\node_modules\ejs\lib\ejs.js:662:17)
    at tryHandleCache (c:\git\OctoFarm\node_modules\ejs\lib\ejs.js:255:36)
    at View.exports.renderFile [as engine] (c:\git\OctoFarm\node_modules\ejs\lib\ejs.js:461:10)
    at View.render (c:\git\OctoFarm\node_modules\express\lib\view.js:135:8)
    at tryRender (c:\git\OctoFarm\node_modules\express\lib\application.js:640:10)
    at Function.render (c:\git\OctoFarm\node_modules\express\lib\application.js:592:3)
    at ServerResponse.render (c:\git\OctoFarm\node_modules\express\lib\response.js:1012:7)
    at ServerResponse.res.render (c:\git\OctoFarm\node_modules\express-ejs-layouts\lib\express-layouts.js:77:18)
    at c:\git\OctoFarm\routes\index.js:87:7
    at processTicksAndRejections (internal/process/task_queues.js:97:5)

NotExpectedYet, 2020-04-06:
git pull, 

That line 70 doesn't exist anymore as it's now:
      <% if(selectedProfile != null && typeof printers[sorted.actualIndex].profiles[selectedProfile] != 'undefined'){ %>

NotExpectedYet, 2020-04-06:
You can confirm with printer manager too as you should be able to open that up now with the non-vanilla machine

TheyCallMeZ, 2020-04-06:
Yep I can open the file manager without crash now.
and the info screen comes up for Canvas Hub, but strangely not the PrusaPrint still... I will dabble a bit to see if I can't sort that out for you. 
[image]

(clicking on the gear brings up the last printer I looked at, so if it was the S5, it brings that up, if it was the canvas hub it brings that up) But as far as this particular issue goes it's resolved!

closed 2020-04-06 by TheyCallMeZ

NotExpectedYet, 2020-04-06:
If you can show me error logs for the prusa I can try to help :) 

I'll take a look at the modal issue as that's a regression... cheers for pointing that out!

NotExpectedYet, 2020-04-06:
Yeah don't have the modal issue, strange...

TheyCallMeZ, 2020-04-06:
this may be a silly question, but are you looking for /dev/ttyUSB? when
searching out the port?
The PrusaPrint image is a pinheader directly connecting the two boards.
thus uses a Serial port of /dev/ttyAMA0

NotExpectedYet, 2020-04-06:
The ports are pulled, I don't look for a specific port. Your port lists pull as a indexed list and I just match the index to either selectedPort/currentPort on OctoPrint.

TheyCallMeZ, 2020-04-06:
The one error that I do see in my debug console is as follows

TheyCallMeZ, 2020-04-06:
Grabbed system for Printer: 0
null: StringifyError {name: "Error", message: "Circular Structure Detected"}
message: "Circular Structure Detected"
name: "Error"
__proto__: Object {constructor: }
constructor: function StringifyError(m) { … }
[[FunctionLocation]]: internal#location
[[Scopes]]: Scopes[2]
arguments: TypeError: 'caller', 'callee', and 'arguments' properties may
not be accessed on strict mode functions or the arguments objects for calls
to them
caller: TypeError: 'caller', 'callee', and 'arguments' properties may not
be accessed on strict mode functions or the arguments objects for calls to
them
length: 1
name: "StringifyError"
prototype: Object {constructor: }
__proto__: function () { … }
__proto__: Object {constructor: , __defineGetter__: , __defineSetter__: , …}
Grabbed system for Printer: 0
Grabbed state for Printer: 0

>

NotExpectedYet, 2020-04-06:
It looks like it's having issues grabbing some information on the initial checks. That's certainly something I'd have to have one handy to check I think and test against because that's triggering an error on the initial check after opening the websocket connection. 

Can I use a PiZero or a Pi to fake a canvas hub and attach it to one of my instances? Or would I need the real thing?

TheyCallMeZ, 2020-04-06:
Yes you can, Mosaic has an article about it
https://support.mosaicmfg.com/hc/en-us/articles/360012423734-Setup-Guide-DIY-CANVAS-Hub-OctoPi-CANVAS-and-P2-Plugins-

With Regard to the PrusaPrint one, it looks like they have given up on
their premade image so I am going to reflash that with stock octoprint and
see how that goes.
(for reference their directions I will be following)
https://help.prusa3d.com/en/article/octoprint-building-an-image-for-raspberry-pi-zero-w_2182#2_how_to_download_or_create_your_own_r_pi_zero_w_octo_print_image

NotExpectedYet, 2020-04-06:
Sweet thanks, and this would replicate your setup? 

Do me a favour and create another issue log for this one please, and when I've got chance I'll get it setup myself and see what's going on :) hopefully it's something simple, the majority of stuff isn't very dependant on most of that information so shouldn't be too hard to at least put a check in for it once I figure out what it's not finding properly. 

I gather there were no "Grabbing profiles for Printer:0" status for those in the logs?

TheyCallMeZ, 2020-04-06:
Yes that would duplicate my setup and the canvas hub was number 2 in the
list. I'll get a issue opened shortly

commit ab148f94 (2020-04-06): Fixed: #22