OctoFarm/OctoFarm #23: Camera doesn't apply multiple rotation settings

NotExpectedYet, 2020-04-06:
Camera currently only applies the top rotational setting and doesn't apply both.

NotExpectedYet, 2020-04-09:
Just on views from my tests

dburr, 2020-04-12:
In my case, "Camera currently only applies the top rotational setting" does not seem to apply. I have my RasPi camera mounted upside down because of cable management issues, and so I have to have both "flip horizontally" and "flip vertically" enabled. However OctoFarm shows the image correctly flipped vertically, but NOT flipped horizontally as it should be (so, in effect, the image appears mirrored.) So in my case it's the SECOND setting (flip vertically) that is being correctly applied, NOT the first (flip horizontally.)
[image]
[image]

NotExpectedYet, 2020-04-13:
The title doesn't say that, I'd edited it... As it says it applies the first, not multiple. Will be fixed in a later release.

dburr, 2020-04-13:
> The title doesn't say that, I'd edited it... As it says it applies the first, not multiple. Will be fixed in a later release. Just because it's first there doesn't mean it's first in the code as well. I know where the bug is but thanks for the additional info.

OK, thanks for the reply. Looking forward to it -- viewing my (printer's) world horizontally flipped is a bit disorientating ;-)

NotExpectedYet, 2020-04-13:
No worries :) Yeah can imagine, I got lazy when I was doing the views if I'm honest. I remember thinking, no one will need both aha. Certainly be sorted for the release of 1.1.5 though. I don't have any work arounds at the minute either but I'll see if I can push that to master within the week for you as  a few other people have mentioned it.

NotExpectedYet, 2020-04-28:
@dburr 1.1.5 is on development currently and fixes that. You can check the changelog to see if you want to change over. Currently there's some stat's missing but everything else is in there.

closed 2020-06-04 by NotExpectedYet

jcassar, 2021-09-08:
Hello, does this behavior was resolved? I am running OctoFarm 1.1.13-hotfix and OctoPrint Version 1.6.1 and got the same issues when Flip webcam horizontally and vertically are selected in Webcam & Timelapse panel. Any ideas to have the world horizontally flipped. Thanks!

jcassar, 2021-09-08:
No worries, restarting octoprint fix that.