OctoFarm/OctoFarm #24: Canvas Hub Machine Info not showing

TheyCallMeZ, 2020-04-06:
Occasionally (more often than not)
the print manager will show up empty

[image]

but if you go to a valid machine and then back to this machine, it brings up the information for the previous machine.

NotExpectedYet, 2020-04-10:
There's a push coming down the line shortly once I've had some confirmation the fixes work for prusa, I'm wondering if the canvas is producing a similar issue. 

Will be done as a release shortly, be sure to check and let me know :)

NotExpectedYet, 2020-08-10:
I'm going to close this issue as I've had reports from other people with canvas hub it's working. If it isn't the case please feel free to re-open.

closed 2020-08-10 by NotExpectedYet