OctoFarm/OctoFarm #26: Printer information not retrieved (completely)

TheLion, 2020-04-09:
Hi,

Installed and configured OctoFarm without issues for my two printers, running on two separate Octoprint RPi's.

Strange thing is that for the Sidewinder X1 all information is retrieved and populated, but for the other (Original Prusa i3 MK3S), not all information is retrieved, like the name. But some information, like that it is printing or idle and the temperatures, is retrieved.

See screenshots for more info.

[image]

[image]

[image]

What can cause this behaviour ?

NotExpectedYet, 2020-04-09:
Have you enabled CORS and restarted the Pi that is attached?

TheLion, 2020-04-09:
Yup, I did. Even disabled CORS, restarted the Pi, re-enabled CORS and restarted again, no change.
Octoprint version on both systems is 1.4.0

NotExpectedYet, 2020-04-09:
Could you try the latest development version? 

If you let me know how you've installed it I can instruct how to switch. There are some improvements to the logging and websocket information.

TheLion, 2020-04-09:
I used the production installation section to install OctoFarm, if that is what you mean.

NotExpectedYet, 2020-04-09:
Oh, ok... easy enough. 

Make sure you stop the current running version. 

cd into the OctoFarm directory, then run the following commands. 

`git checkout development`
`git pull`

Then run
`npm update`
`npm install`
`npm start`

That will move you over to the development branch, I've been holding off on merging the two currently but I think I might do an early release because a few more people are having this issue with the old implementation. 

The latest changelog for that so far is here. 
https://github.com/NotExpectedYet/OctoFarm/blob/development/CHANGELOG.md

TheLion, 2020-04-09:
When doing git pull after switching to development, it reports it is already up to date. OctoFarm version shows 1.1.4.4 in the webgui, didn't check version before the switch ;-)
It did show message below when switching to development

`M       config/db.js`
`M       package-lock.json`
`Branch 'development' set up to track remote branch 'development' from 'origin'.`
`Switched to a new branch 'development'`

NotExpectedYet, 2020-04-09:
Yeah your on the latest, has that improved things? 
It's just saying "i hope" It's skipping those files, but if you've got 1.1.4.4 your on the latest commit of that branch :)

Crowlord, 2020-04-09:
I am also still getting this. 

My Kossel populates but both my prusas are nada. It does read the nozzle size etc now though.

None of them are able to have their files read. I can even upload and it will instantly vanish. 

Just updated to dev branch no improvement. 

It also could not auto determine the webcam paths, I had to remove the printers and readd them with full details as you can get into them to adjust it. 

I have even tried wiping mongodb and starting again.

Crowlord, 2020-04-09:
It does feel like a prusa issue as i noticed the orange on another report as well!

TheLion, 2020-04-09:
Ah. Thought that something went wrong ;-)

No, it doesn't change anything unfortunately.

Anything, like logfiles, I can provide you with? Or check anything?

NotExpectedYet, 2020-04-09:
Yes, look in OctoFarm/Logs you'll see everything there. 

If your both on the dev branch and it's your prusa machines that are not connecting to the websocket, and API checks seem to be failing. 

What is the OctoPrint image? Is it the prusa one or the official one? 

If you can pass me the OctoFarm logs, production.log and productionError.log I'll take a look and try to advice. 

Please note though, if it's the prusa OctoPrint image that isn't supported.

TheLion, 2020-04-09:
I decided to install OctoFarm due to a video from Chris's Basement yesterday. He also uses Prusa printers, but not sure which ones.

I use the official Octoprint images on both printers, not the Prusa one.

The productionError.log is empty. The production.log is attached to this message. Other files are not in the logs folder.

[production.log](https://github.com/NotExpectedYet/OctoFarm/files/4457048/production.log)

TheLion, 2020-04-09:
Oh, are there any plugins that are known to interfere? Even though both Octoprint's run almost identical installs and plugins.

Crowlord, 2020-04-09:
[production.log](https://github.com/NotExpectedYet/OctoFarm/files/4457094/production.log)
[productionError.log](https://github.com/NotExpectedYet/OctoFarm/files/4457095/productionError.log)

(One printer is offline hence readings in error

I do note that TheLion actually has some ON OPENSearching... lines that are missing from mine. 

Just to confirm all stock octoprint, no change in safe mode.

NotExpectedYet, 2020-04-09:
OpenLion
I think git has mangled your versions... There's a part of that production.log that's got an old console log in it. 

Could you try: 
Stop the server
`git fetch --all`
`git reset --hard origin/developement`

Then `npm start` again
Then if your still having issues forward me the logs again please. Sorry, dunno why git pull didn't pull it all -_-

NotExpectedYet, 2020-04-09:
Crowlord (Why are @ not working -_- )

Your's is timing out on connecting on one of the API connections causing those errors, what's your network like? WiFi? 

You any good with an text editor like notepad++ ? 

I can show you how to increase the timeout for now to see if it helps grab the information.

NotExpectedYet, 2020-04-09:
/OctoFarm/runners/state.js
Line 235 you'll see the following code: 
`setTimeout(() => reject(new Error("timeout")), 1000)`
Change the 1000 to something like 3000, it's in milliseconds and see if that helps your situation.

Crowlord, 2020-04-09:
I will give it a tweak. The printer in question is offline in octoprint as I forgot to connect it but online on the network (I am printing from SD as its a kossel and mangles Local prints)

NotExpectedYet, 2020-04-09:
> I will give it a tweak. The printer in question is offline in octoprint as I forgot to connect it but online on the network (I am printing from SD as its a kossel and mangles Local prints)

I don't think printing from SD card should effect anything but that will certainly explain why your not seeing any files. I currently only support local which has been the case for ages and nobody ever requested it. I don't think OctoPrint changes anything else in the background but basically your API calls are not grabbing the settings and something else which would explain why the Printer Manager isn't displaying. It needs more checks in that regard I'll be honest.

Crowlord, 2020-04-09:
That worked for the Kossel. (printer 1) I am now seeing it properly and its files for the first time. 

The prusas (0 and 2) are still invisible. 

Re the SD thingy. I do have local files too but I switch to SD if its a curvy file as Octo sucks with curves on it. 

TLDR: The timeout fix has sorted out the files issiue on one printer but the prusa issues remain so I am now even with @TheLion

Crowlord, 2020-04-09:
[image]

NotExpectedYet, 2020-04-09:
Getting somewhere at least! So, next release I'll work in a timeout increase for you so you can do it via the UI and it won't overwrite each time. 

The Prusa I am seriously confused as to why that doesn't work. Especially in safe mode as that would completely rule out your settings/profile issues. Which

Crowlord, 2020-04-09:
I do have very large file systems on them. My "housekeeping" is poor. 

I need to do some purging so fun easter project... could it be as simple as a timeout!

TheLion, 2020-04-09:
> I think git has mangled your versions... There's a part of that production.log that's got an old console log in it.

I think it was an old entry in the log.
After stopping and the git commands, the logfile isn't even created now anymore and OctoFarm runs fine, except for the Prusa.

NotExpectedYet, 2020-04-09:
> > I think git has mangled your versions... There's a part of that production.log that's got an old console log in it.
> 
> I think it was an old entry in the log.
> After stopping and the git commands, the logfile isn't even created now anymore and OctoFarm runs fine, except for the Prusa.

The entry logs are re-written each time, they should just replace the old file until I put in a more robust system. 

> I do have very large file systems on them. My "housekeeping" is poor.
> 
> I need to do some purging so fun easter project... could it be as simple as a timeout!

Ahh that might explain it, I've never really pushed the limits of the file system checking but regardless of size OctoPrint should still push back that information and it shouldn't time out as the connection would be active straight away. 

I'll do some tests on that eventuality for you, but it was the same in 1.0 and I've had people who literally have their SD cards full. 

Colour me really confused with the Prusa's guys, I'm sorry to say. I currently have no way of plugging one in and testing it, unless you could send me through and copy of your Pi's SD cards to test that on, as I sincerely doubt it's anything to do with the specific printer.

NotExpectedYet, 2020-04-09:
How far did you push the timeout's btw? 

You could try maxing out to 10000ms and see if it still persists.

Crowlord, 2020-04-09:
timeout set to 10000 and it has pulled the printers names etc!

I did just delete some 70mb+ gcodes but there is definitely a difference. 

Still waiting to pull some files from Rosie but it appeared to work!

I now have all printers names and 2 lots of the local files on system.

I have full controls settings etc where i had nothing. 

[image]

NotExpectedYet, 2020-04-09:
Right, so that's something I'll test and push fixes for. I've got an idea, if I make some changes and push to master would you mind testing them you two? 

See if we can't sort this :)

Crowlord, 2020-04-09:
dont mind at all

NotExpectedYet, 2020-04-09:
Right update to master and see if that helps. 

I've not touched the timeout, I've just made it not wait for the file's on the initial scan before continuing. If this does help, I'll come up with a more elegant solution when I've more time.

TheLion, 2020-04-09:
No problem at all.

NotExpectedYet, 2020-04-09:
Just make sure it's development not master ahaha. Hopefully this will allow the scanning of the settings and profiles to continue and not need a timeout increase.

NotExpectedYet, 2020-04-09:
What usage did you both have on SD cards?

TheLion, 2020-04-09:
> 
> 
> Just make sure it's development not master like I'd originally typed ahaha. Hopefully this will allow the scanning of the settings and profiles to continue and not need a timeout increase.

No change unfortunately :-(

TheLion, 2020-04-09:
> 
> 
> What usage did you both have on SD cards?

Not using SD cards.

NotExpectedYet, 2020-04-09:
I thought you were running it on a Pi? I'm meaning file storage. 

Damn... Right I'll try and investigate further and see what I can come up with. 

Did you try the timeout change TheLion?

NotExpectedYet, 2020-04-09:
Crowlord, if that push doesn't help then and the timeout did I will work in a customise option for that for the next release. For now, just edit manually and hopefully that should keep you golden. The timeout is on the API so it doesn't effect any of the websocket connections with the information.

Crowlord, 2020-04-09:
I am getting an error on  filemanager page with that push 

`TypeError: D:\octofarm\views\filemanager.ejs:48
    46|         </center>

    47|         <div id="printerList" class="list-group">

 >> 48|           <%- include('./partials/FileManager/printerList.ejs') %>

    49|         </div>

    50|       </div>

    51|       <div class="card-footer fileFooter text-muted"></div>

D:\octofarm\views\partials\FileManager\printerList.ejs:71
    69|         </div>

    70|         

 >> 71|         <% if(selectedProfile != null && typeof printers[sorted.actualIndex].profiles[selectedProfile] != 'undefined'){ %>

    72|           <small class="pt-2 float-left"

    73|           ><i class="fas fa-cube"></i> <b>H:</b> <%=

    74|           printers[sorted.actualIndex].profiles[selectedProfile].volume.height%>mm x <b>W:</b> <%=

Cannot read property '_default' of undefined
    at eval (D:\octofarm\views\partials\FileManager\printerList.ejs:71:82)
    at Array.forEach (<anonymous>)
    at eval (D:\octofarm\views\partials\FileManager\printerList.ejs:8:20)
    at printerList (D:\octofarm\node_modules\ejs\lib\ejs.js:662:17)
    at include (D:\octofarm\node_modules\ejs\lib\ejs.js:660:39)
    at eval (D:\octofarm\views\filemanager.ejs:11:17)
    at filemanager (D:\octofarm\node_modules\ejs\lib\ejs.js:662:17)
    at tryHandleCache (D:\octofarm\node_modules\ejs\lib\ejs.js:255:36)
    at View.exports.renderFile [as engine] (D:\octofarm\node_modules\ejs\lib\ejs.js:461:10)
    at View.render (D:\octofarm\node_modules\express\lib\view.js:135:8)
    at tryRender (D:\octofarm\node_modules\express\lib\application.js:640:10)
    at Function.render (D:\octofarm\node_modules\express\lib\application.js:592:3)
    at ServerResponse.render (D:\octofarm\node_modules\express\lib\response.js:1012:7)
    at ServerResponse.res.render (D:\octofarm\node_modules\express-ejs-layouts\lib\express-layouts.js:77:18)
    at D:\octofarm\routes\index.js:87:7
    at runMicrotasks (<anonymous>)`

TheLion, 2020-04-09:
Octoprint is running on Pi's, OctoFarm not.

Both Pi's start from SD, but run from a 128G USB stick after boot.

One has 4% (3.7G) in use and the other 9% (10G)

Just set the timeout to 10000 and it pulled the information right away!

Don't have the error on filemanager, but it shows information for the first time now

NotExpectedYet, 2020-04-09:
Yeah that's a possible issue with the changes I made. Basically all I've told it to do is not wait for that and allow the connections to continue. 

Yeah, I've reverted the changes and pushed it back. 

Right so I believe then you were both having separate issues. 

TheLion: I believe yours is down to the USB stick, I can only gather that's slowing the responses down some and causing it to timeout. 

Crowlord: Yours seems to be the large amount of files it's trying to pull but I'm just uploading 411 gcodes to my testing instances now to try and replicate but that will take a while. 

Both of you my recommendation is to pull the reverted changes from the development version and then make the timeout edits until I find a more graceful way to handle that. I'll certainly get it reproduced though, and once I've solved that you should both be fine going forward. You can at least use the system now with the changes I've recommended going forward.

Crowlord, 2020-04-09:
Great to help, thanks for the awesome tool.

NotExpectedYet, 2020-04-09:
No trouble, thanks for that little experiment, I'll get it sorted :)

TheLion, 2020-04-09:
> TheLion: I believe yours is down to the USB stick, I can only gather that's slowing the responses down some and causing it to timeout.

The USB sticks are faster than the SD, so that should not be the issue. But I'm glad that we got it sorted so far!

Thanks for the support and the great tool!

NotExpectedYet, 2020-04-09:
> > TheLion: I believe yours is down to the USB stick, I can only gather that's slowing the responses down some and causing it to timeout.
> 
> The USB sticks are faster than the SD, so that should not be the issue. But I'm glad that we got it sorted so far!
> 
> Thanks for the support and the great tool!

Thanks for pointing that out, I was under the assumption the SD bus was better speed wise. Well I've just tried with 119 files on my instance so I gather I'll put it on a Pi as mine is a bit more beefy running on a VM. 

I'll keep this issue open and let you know my progress when I get more chance to test.

TheLion, 2020-04-09:
> Thanks for pointing that out, I was under the assumption the SD bus was better speed wise.

I'm using SSD based USB sticks, so they are crazy fast!

> Well I've just tried with 119 files on my instance so I gather I'll put it on a Pi as mine is a bit more beefy running on a VM.

You got Octoprint running on a VM? How did you get a stable and reliable USB connection to your printer?

> I'll keep this issue open and let you know my progress when I get more chance to test.

Great! Will keep an eye out.

NotExpectedYet, 2020-04-10:
> > Thanks for pointing that out, I was under the assumption the SD bus was better speed wise.
> 
> I'm using SSD based USB sticks, so they are crazy fast!

Oh they are a thing!? ahahaha

> > Well I've just tried with 119 files on my instance so I gather I'll put it on a Pi as mine is a bit more beefy running on a VM.
> 
> You got Octoprint running on a VM? How did you get a stable and reliable USB connection to your printer?

I have two setups currently, one testing server all setup with virtual printers... This runs on a virtual machine on my UnRaid server. The second is my main PC on an i3 that runs 3 instances over PCI USB3.0 port adapter so I could run 8 printers. The discord has a user that uses rfc to send usb serial over network and he uses skinny clients with multiple printers attached and OctoPrint is installed on a VM elseware on his network. He says it works rather well. 

> > I'll keep this issue open and let you know my progress when I get more chance to test.
> 
> Great! Will keep an eye out.

Please pull the latest master, there's a new configuration file with options, but I don't think you'll need them, it automagically retry's the connections now :)

Crowlord, 2020-04-10:
The latest dev is throwing errors for me and a crash 

[production.log](https://github.com/NotExpectedYet/OctoFarm/files/4460793/production.log)
[productionError.log](https://github.com/NotExpectedYet/OctoFarm/files/4460794/productionError.log)
[image]

NotExpectedYet, 2020-04-10:
It hasn't committed the file to dev branch, try pull again ahahahah

Crowlord, 2020-04-10:
Thats loads better.

I would put a warning about long startups with big file systems and it still struggles to (wont) load my printer with 670 files! (Yeah housekeeping)

Printer with 197 has no problems and one with 380  takes an age but does manage it. 

It found that a timeout of 100000 (i think that was the amount)! does it for the 670 but thats ludicrously big. 

You can now tell who all the gcode hoarders are :D

NotExpectedYet, 2020-04-10:
Thanks for the feedback! 

Epic I'll get it pushed to master so you can all get back on the normal branch. This issue will close when that's done :) and add a note to the change log regarding it. 

Yeah it's weird one like, I'll go through the file scanning and see if there are any improvements I can make with it there, this is on how fast the information comes back... so I gather I've got a lot of files to add to my instances to replicate ahaha

closed 2020-04-10 by NotExpectedYet

sonjamichelle, 2020-12-26:
I found that I was having the same websocket problems with my Prusa i3 MK3/S with the current 3.92 firmware and the latest OctoPi build 1.5.2 and the master branch of OctoFarm. So I checked out the development branch updated but continued to have issues. I looked at the logs, saw something that made me do a search for "setting your OctoFarm user to an admin" this led me to this page ( Not sure how, but I'm glad it did) https://octofarm.net/installation. In there it had this  statement:

`Note: since OctoPrint version 1.4.1 it is recommended to connect using the Application Key / User Key detailed below. Versions before that are fine using the Global API Key generated by OctoPrint.` 

SOOOO, I generated an application API for OctoFarm, redid my printer discovery, added my details and the new API, after a couple of resyncs all is well!

Not sure if this would have worked in the master branch or not, but I'm hesitant to go back now that things are working. 

I think I will wait until this issue is closed and we have a definitive answer to the Prusa i3 idiosyncrosies.

NotExpectedYet, 2020-12-26:
This issue is closed? 

Yeah that's correct, OctoPrint updated and the websocket Auth stopped working with the global API key. Gina said she would rather leave this as a feature for later versions as she wants to move away from that key anyway. 

Your actually a few commits behind on development currently. I updated master branch before Christmas with the latest release. You are fine to go back to master, this issue with the API key was resolved some months ago when the initial change with octoprint happened. Hope that helps.

commit e9a08221 (2020-04-10): Uprev 1.1.4.6-bugfix

Fixed #26