OctoFarm/OctoFarm #28: OctoFarm resets feed rates

paukstelis, 2020-04-10:
I've noticed that OctoFarm resets feed rates (and perhaps flow rates?) after each print. Is there a way to alter this behaviour?

NotExpectedYet, 2020-04-10:
It doesn't reset them, it remembers your previously set feed/flow rate and re-applies that. 

Currently no but if you would like that feature please add it to my feathub page.

paukstelis, 2020-04-10:
That assumes the feed/flow changes were made through OctoFarm and not on the printer directly. I think most of the time those changes would be made while examining the print. I'll put in a request.

NotExpectedYet, 2020-04-10:
It does, but if your making the feed/flow rate changes there why not start the print from there?

NotExpectedYet, 2020-04-10:
Gonna close this ticket anyway as it's not a bug. 

Feel free to give me some use cases in your feat-hub :) would rather understand your process to try and get a better fit.

closed 2020-04-10 by NotExpectedYet