OctoFarm/OctoFarm #29: Unable to save new printers: UnhandledPromiseRejectionWarning

MedadRufus, 2020-04-11:
When I try to save new printers via the System>Printers menu, both by manually typing in printers or by uploading the Json file, it throws the error:

```bash

(node:9964) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'close' of undefined
    at Function.reset (/home/medad/OctoFarm/runners/state.js:452:41)
    at /home/medad/OctoFarm/routes/printers.js:71:30
    at Layer.handle [as handle_request] (/home/medad/OctoFarm/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/medad/OctoFarm/node_modules/express/lib/router/route.js:137:13)
    at ensureAuthenticated (/home/medad/OctoFarm/config/auth.js:10:14)
    at Layer.handle [as handle_request] (/home/medad/OctoFarm/node_modules/express/lib/router/layer.js:95:5)
    at next (/home/medad/OctoFarm/node_modules/express/lib/router/route.js:137:13)
    at Route.dispatch (/home/medad/OctoFarm/node_modules/express/lib/router/route.js:112:3)
    at Layer.handle [as handle_request] (/home/medad/OctoFarm/node_modules/express/lib/router/layer.js:95:5)
    at /home/medad/OctoFarm/node_modules/express/lib/router/index.js:281:22
    at Function.process_params (/home/medad/OctoFarm/node_modules/express/lib/router/index.js:335:12)
    at next (/home/medad/OctoFarm/node_modules/express/lib/router/index.js:275:10)
    at Function.handle (/home/medad/OctoFarm/node_modules/express/lib/router/index.js:174:3)
    at router (/home/medad/OctoFarm/node_modules/express/lib/router/index.js:47:12)
    at Layer.handle [as handle_request] (/home/medad/OctoFarm/node_modules/express/lib/router/layer.js:95:5)
    at trim_prefix (/home/medad/OctoFarm/node_modules/express/lib/router/index.js:317:13)
(node:9964) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
(node:9964) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```
Just before the above errors, I had these errors to do with connecting to the printers. I did not have them online while testing so it is expected.

```bash

Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Retrying Connection with: 10000
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Couldn't grab initial connection for Printer: 0 Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)

```

I am running the latest code on master. I wonder what is going on.

MedadRufus, 2020-04-11:
I could all the first few printers in the interface that appears right after making a new account. But I cannot add more later.

NotExpectedYet, 2020-04-11:
Pulll latest, was an oversight on the reset after some runner changes

MedadRufus, 2020-04-11:
Looks like it works. Thanks

closed 2020-04-11 by MedadRufus