OctoFarm/OctoFarm #30: Long wait times after adding in new list of printers

MedadRufus, 2020-04-11:
Octofarm tries to connect to all printers as soon as new list of printers uploaded and takes a very very long time to do that.

I do not have the Octoprint servers running while I add in the new printers using the system>printers interface. Octofarm tries to connect to all of them and times out after a while. If I add a lot of printers, it takes a long long time to time out all of them, locking up the user interface. I think Octofarm should not try to connect to all the octoprint instances immediately after the details are added in, instead just storing the IP address and API key, and connect later when the User requests to do so.

Log from bash:
```bash

Grabbed 5 for checking
Setting up Statistics Collection
Statistics collection has started...
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Couldn't grab initial connection for Printer: 0 Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Couldn't grab initial connection for Printer: 1 Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Couldn't grab initial connection for Printer: 2 Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Couldn't grab initial connection for Printer: 3 Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
Couldn't grab initial connection for Printer: 4 Error: timeout
    at Timeout._onTimeout (/home/medad/OctoFarm/runners/state.js:247:37)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)
```

NotExpectedYet, 2020-04-12:
That should be better in a future release, but I've got users who want automation, and users who don't currently. The easiest way to get around this is add all your printers to the database and keep them there, you don't have to save per printer. I'm going to close this as it's not a bug but intended, but please feel free to drop a feature request in for something if you think it should work better in someway or I'm happy to discuss further. The second timeout was added due to prusa printers, and some people with ungodly amounts of files on their systems... in the 400+ range. How often are your restarting your server? because that's intended to run constantly too and once the initial scan for printers has been made there should be no reason apart from updates to restart it.

closed 2020-04-12 by NotExpectedYet

MedadRufus, 2020-04-13:
Thanks for the update