OctoFarm/OctoFarm #31: Feature Request - ability to add unmanaged camera streams to Panel & Camera Views

dburr, 2020-04-13:
In addition to my FDM printers, I have several resin printers that I run, that are NOT managed by Octoprint, but they do have webcams pointed at them. It would be nice to be able to add camera viewer windows for them to the Panel & Camera View screens (even though I wouldn't be able to control them from Octofarm) just so I can keep an eye on them, see when print jobs are done or if they're in trouble, etc.

NotExpectedYet, 2020-04-13:
Closing, feature requests are on Feathub please: https://feathub.com/NotExpectedYet/Octo

closed 2020-04-13 by NotExpectedYet

dburr, 2020-04-13:
> Closing, feature requests are on Feathub please: https://feathub.com/NotExpectedYet/Octo

Apologies, I was not aware of feathub. Will repost there.

NotExpectedYet, 2020-04-13:
No trouble :) it's mentioned on the readme for future reference.

NotExpectedYet, 2020-04-13:
I gave it an updoot too as I have been thinking about that feature myself. Just unsure how to go about it at the moment.