OctoFarm/OctoFarm #32: Cannot reach OctoFarm using 'Installation Production' Documentation

adamsiem, 2020-04-13:
I followed the documentation for 'Installation Production > Debian/Linux Specific', but cannot access OctoFarm on my local network at port 4000.

NOTE: This is on Raspbian, and I have Pi-Hole running on the same device, successfully, at the IP address only, i.e. no custom port for the web-server.

**It started... woohoo!**

pi@raspberrypi:~/OctoFarm $ hostname -I
192.168.1.190
pi@raspberrypi:~/OctoFarm $ npm start

> octofarm@1.1.4 start /home/pi/OctoFarm
> node app.js > logs/production.log 2> logs/productionError.log

**The productionError.log is empty... woohoo!**

GNU nano 3.2 productionError.log

**My config looks... good...?**

GNU nano 3.2             config/db.js

module.exports = {
  MongoURI: "mongodb://127.0.0.1:27017/octofarm"
};
//Example Local URL: "mongodb://192.168.1.5:27017/octofarm"
//Example Remote URL: "mongodb+srv://s1mpleman:<YOUR PASSWORD>$

This resembles: https://github.com/NotExpectedYet/OctoFarm/issues/16

NotExpectedYet, 2020-04-13:
PiHole image uses Rasbian so there's no MongoDB support for the version that's required. 

Your config looks fine, but you'd have to use an External MongoDB installation to get it to work. I gather it's not starting because it's not able to connect to the local MongoDB instance. Sorry there isn't more loggin but it usually produces are error for that although I've not specifically tried it. 

You could also setup the 64-bit Ubuntu version on the Pi and install both OctoFarm + PiHole as a dependancy on that if you still wanted to keep your Pi for those.

gmccauley, 2020-04-13:
+ 1 on ubuntu.  I used an old Pi 3B I had laying around and installed the 64 bit version of ubuntu and it's working great.

adamsiem, 2020-04-15:
I confirmed this is resolved on Ubuntu Server 64-bit 18.04 LTS on a Raspberry Pi 4 B. Nice work!

closed 2020-04-15 by adamsiem