OctoFarm/OctoFarm #33: Full Screen Camera

gmccauley, 2020-04-13:
When double clicking on the camera image from the Panel or Camera views (to show the camera stream full screen), the bottom 1/4rd of the stream is cut off.

I assume that the stream is trying to maintain aspect ration and since my monitors aren't perfect squares, the bottom section gets cut off.

I've attached some pics to help explain.

[image]
[image]

NotExpectedYet, 2020-04-13:
Cheers for that, did wonder if it cut off 4:3 but I've none to test with. I'll have a look about maintaining aspect ratio, html is annoying for sizing by height but I'm sure I can work something out.

gmccauley, 2020-04-28:
Confirmed Fixed in 1.1.4.9-dev-1 release

NotExpectedYet, 2020-06-04:
Fixed in recent 1.1.5 push

closed 2020-06-04 by NotExpectedYet