OctoFarm/OctoFarm #34: display date times in local timezone (or at least option to do so)

tonysurma, 2020-04-16:
I cannot find a setting for this and it appears that all datetimes are displayed in GMT?  For example my currently running prints started 'in the future' from my perspective.

Thanks for this great software

NotExpectedYet, 2020-04-16:
Hey thanks :) 

It uses a JavaScript function to generate that, what browser are you using as it should just generate it in ISO format for your locale.

tonysurma, 2020-04-16:
I have seen this on Brave, Edge (Chromium) stable and Edge (Chromium) Beta on both windows PC and Android Phone.

NotExpectedYet, 2020-04-16:
Hmmm, I'm not 100% sure on that one but thank your for the thorough checking. 

What is your locale? I'll have to do some research into that specific JavaScript function as I've not had anyone else mention it since loading that feature and there's a few users from different locales.

tonysurma, 2020-04-16:
I am in Chicago Central Time.

If you have a pointer to where it is in the code I can test and play with it to see if I can isolate as well.

NotExpectedYet, 2020-04-16:
Cool thank you I'll see if I can dig anything up as well. 

It's in:
/views/js/lib/modules/currentOperations.js 
`
    let currentDate = new Date(); 
    currentDate = currentDate.getTime(); 
    let futureDateString = new Date(currentDate + current.timeRemaining * 1000).toDateString()
    let futureTimeString = new Date(currentDate + current.timeRemaining * 1000).toTimeString()
    futureTimeString = futureTimeString.substring(0, 8);
    let dateComplete = futureDateString + ": " + futureTimeString;
`

gmccauley, 2020-04-16:
Where do you have OctoFarm instlaled?  I have it installed on a Pi and had the same problem until I changed the timezone on the Pi to match my current timezone.

NotExpectedYet, 2020-04-16:
That would do it considering nodejs will use the systems local locale.

tonysurma, 2020-04-16:
Ah so it is a server side determination.  I was hoping client.  It is installed via docker on my synology NAS.  The NAS has the right timezone but of course through docker doesn't set the timezone.  I will try the TZ environment variable via docker and a few other approaches and see if that makes a difference.

Thanks for the rapid engagement on this.

tonysurma, 2020-04-16:
bingo - TZ variable did it.

Maybe we can add to installation guidelines for docker

tonysurma, 2020-04-16:
One thing I will add is I noticed the history was apparently stored in that 'local' datetime (not as GMT and then rendered in current timezone).

Therefore now that I changed timezones they are not updated.  I don't know that anyone will ever run into this issue but just an FYI

gmccauley, 2020-04-16:
> One thing I will add is I noticed the history was apparently stored in that 'local' datetime (not as GMT and then rendered in current timezone).
> 
> Therefore now that I changed timezones they are not updated. I don't know that anyone will ever run into this issue but just an FYI

Yeah, after I changed timezones, I noticed that the times in the history didn't change as well.

NotExpectedYet, 2020-04-16:
> One thing I will add is I noticed the history was apparently stored in that 'local' datetime (not as GMT and then rendered in current timezone).
> 
> Therefore now that I changed timezones they are not updated. I don't know that anyone will ever run into this issue but just an FYI
The history times are stored in a similar fashion so that would explain that. If you explain to me how you updated the timezone for Docker I'll make a note of it on the read me :) Most of that stuff is donated as I've no experience with it myself.

tonysurma, 2020-04-16:
> The history times are stored in a similar fashion so that would explain that. If you explain to me how you updated the timezone for Docker I'll make a note of it on the read me :) Most of that stuff is donated as I've no experience with it myself.

I'll do a pull request to the readme.md with the changes.  Thanks again

NotExpectedYet, 2020-04-16:
Thanks :) Gonna close now

closed 2020-04-16 by NotExpectedYet