OctoFarm/OctoFarm #36: All full URL for printer IP - Multi Instance w/ haproxy

zdesignstudio, 2020-04-16:
When using HA proxy to append urls to the single PI IP address, OctoFarm doesnt allow the additional /name to be added, only IP address.

For example, instead of accessing a multi instance printer as 192.168.1.10:5000 or 5001 you can do 192.168.1.10/printer1, printer2, cr10, etc.
If haproxy is setup this way it does not allow access via port directly so you are unable to add the device in OctoFarm.

NotExpectedYet, 2020-04-16:
feature requests are handled on the feathub page please. 

it's already been noted and requested please go add your vote there.

closed 2020-04-16 by NotExpectedYet

zdesignstudio, 2020-04-16:
Apologies.
Didnt know that site was still used. Im guessing they are down again as I am unable to reach it.

NotExpectedYet, 2020-04-16:
yeah I'm looking to move in due course to a better platform. 

your feature is been added for 1.1.5 though :) 

It seems to be up but there SSL isnt working currently.

zdesignstudio, 2020-04-16:
Awesome.

Thanks for following up James.

NotExpectedYet, 2020-04-16:
no trouble :) 

it's an often requested feature that was never in the original scope. I've re-planned the entire mechanism and I'm hoping to have that released in 2 weeks or so.

NotExpectedYet, 2020-04-17:
https://features.octofarm.net is now live :) 

Please go add your feature requests there, thanks!