OctoFarm/OctoFarm #37: Allow hostnames instead of only IP addresses for machines

mfberry, 2020-04-17:
Overall a minor problem, but I track my printers better through DNS than static IP addresses.

NotExpectedYet, 2020-04-17:
Feature requests are done on feathub... this is already requested. Closing.

closed 2020-04-17 by NotExpectedYet

NotExpectedYet, 2020-04-17:
I've just realised the site is still down -_-

I'm setting up an alternative platform shortly, but your request is coming in 1.1.5

mfberry, 2020-04-17:
Thank you!

I thought I was a crazy person.

NotExpectedYet, 2020-04-17:
No worries, I thought they were just having SSL issues but seems the service is completely down now. Nothing on there twitter page either -_- 

I have lost a lot of blumming feature requests due to that ahaha

NotExpectedYet, 2020-04-17:
I'm just waiting for google to Auth my domain shizzle and it will be live :)

NotExpectedYet, 2020-04-17:
https://features.octofarm.net is now live :) 

Please go add your feature requests there, thanks!