OctoFarm/OctoFarm #38: Can't add more printers at later stage.

kristapsdravnieks, 2020-04-18:
I can't seem to find a way to add more octoprint instances to farm once inital setup wizzard is closed.
is there some url I could re-run setup wizzard or add another octopritn instance trough cli?

NotExpectedYet, 2020-04-18:
Your going to have to explain that one more? 

System -> Printers will let you add more printers and OctoFarm doesn't really have a setup wizard.

kristapsdravnieks, 2020-04-18:
[image]

I don't see printers option under System menu

NotExpectedYet, 2020-04-18:
You need to login as the first user you've created, not the second. The first user defaults to Admin which is the only user who can see that option.

kristapsdravnieks, 2020-04-18:
is there any a way to reset user database?

NotExpectedYet, 2020-04-19:
You'd have to delete that specific table and restart OctoFarm. 

You can do this with the MongoCLI with
`db.users.drop() `

users is the collection, that will save the rest of your data and make you create a new login on restart. 

The easiest way would be to export your printers, and change the db url on OctoFarm, then re-import but you'll loose any collected history.

NotExpectedYet, 2020-04-19:
I'll make more notes about this when I write up some documentation. Been holding off as a lot is changing still going forward.

kristapsdravnieks, 2020-04-19:
Thanks! That worked.

NotExpectedYet, 2020-04-19:
Awesome! 

Will close :)

closed 2020-04-19 by NotExpectedYet