OctoFarm/OctoFarm #39: prints not logging history sometimes

tonysurma, 2020-04-19:
I know this is a bad report when I have to use the word sometimes but out of 31 prints - 3 of them are no where to be found on the history page.

I ensured after the first time it happened (two days ago) that I did all operations through octofarm (as opposed to octopi etc.) - even though I was pretty much 100% doing that already - just to make sure but it didn't make a difference in the two that haven't logged since.

It has logged some so its definitely able to log history, just not sure why these few aren't logging.

Any insight into how to debug this?  Thanks

NotExpectedYet, 2020-04-20:
Hello
Firstly it doesn't matter where the prints are started, OctoFarm uses the Websocket push events from OctoPrint so they can be started from another OctoPrint app if chooses and as long as my app is online it should catch them. I know this isn't very clear just thought I'd explain.  

There's a bug I've been looking into for a while now but never officially wrote it up. I'm not sure myself it's something to do with it not grabbing some tool information from the time of logging. 

Could you give me your OctoPrint version please? 

Before 1.1.5 comes out I shall be trying to reproduce it more and I'll keep track here incase anyone else notices it. Thanks for reporting it.

tonysurma, 2020-04-20:
Thanks I had assumed I wouldn't need to do it via the octofarm UI but wanted to try to reduce variables to see if there was a pattern.

My OctoPrint version is 1.4.0.

I will continue to to try to find a pattern but so far none.  And just to say it was 100% running during the time.

NotExpectedYet, 2020-04-28:
@tonysurma I've done more investigating into this and it seems randomly the printer information it uses is not passed at the time of logging the history record. It's my statistics that create that too as octoprints event doesn't fire the majority of the information it grabs. 1.1.5 has a lot better logs in this regards though and is currently on development branch if you'd like to give this a try. Be good to get your logs from it, but I'm currently going through mine as well. Thankfully I can fire 60 prints off at once aha.

commit 6a282480 (2020-05-04): Fixes for history not collecting data sometimes. Fixed #39
closed 2020-06-04 by NotExpectedYet