OctoFarm/OctoFarm #41: Inaccurate ‘lost connection’ error when app switching on iOS

jw59, 2020-04-22:
If you have OF open in Safari then switch to another app, then back, the client says you’ve lost server connections and lingers for 10 seconds. If you look at the data, this isn’t true it starts to come back sooner. However the page still refreshes after 10 seconds.  Note that camera feeds don’t come back until after the refresh. 

I think it’d be better to show the error after 10 seconds of trying to reconnect. Ie ‘it’s taking longer than usual to connect...’ 

Would be less jarring but have the same effect. 
 
I have a video if needed, can’t seem to attach anything but png here from mobile.

NotExpectedYet, 2020-04-23:
Don't have any access to Safari to test with so for now this won't be fixed sorry.

NotExpectedYet, 2020-04-23:
I gather it's something to do with how iOS implements Server Side Events, what does Chrome do on the same device?

jw59, 2020-04-23:
Chrome does the same thing

NotExpectedYet, 2020-04-23:
If I ever get a device to test on I'll look into it, but I've got no Mac system currently to even get a dev console up on an apple device. There's some changes coming to that area in 1.1.5 so hopefully they might resolve the issue.

jw59, 2020-04-23:
Just for clarity, this isn't on Mac.  It works fine on Mac, this is iOS.

NotExpectedYet, 2020-04-23:
Yeah to develop or at least get web app console logs back from an iPhone I'd need a Mac system to connect the device to and grab logs aha. Sorry for not been clear, I need more coffee.

NotExpectedYet, 2020-08-09:
I've been told latest dev doesn't show this issue. Just trying to confirm now.

NotExpectedYet, 2020-08-17:
Hello, I'm going to close this ticket for now as I've had reports this should be a lot better. Please update and reopen the ticket if it's not the case. thank you.

closed 2020-08-17 by NotExpectedYet