OctoFarm/OctoFarm #42: History page not displaying after removing filament roll

tvi675, 2020-04-27:
After removing a filament roll from the filament manager, the history page now doesn't display. Shows error page with the following:

`TypeError: /home/ubuntu/OctoFarm/views/history.ejs:57
    55|             <% if(typeof print.printHistory.filamentSelection != 'undefined' &&
    56|             print.printHistory.filamentSelection != "None chosen..."){%>
 >> 57|               <% let
    58|             filament = print.printHistory.filamentSelection.roll %> <%=
    59|             filament.name %> [<%= filament.colour %> / <%= filament.type[1] %>]
    60|             <%}else{%> None selected... <% } %>

Cannot read property 'roll' of null
    at eval (/home/ubuntu/OctoFarm/views/history.ejs:60:61)
    at Array.forEach (<anonymous>)
    at eval (/home/ubuntu/OctoFarm/views/history.ejs:13:16)
    at history (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:662:17)
    at tryHandleCache (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:255:36)
    at View.exports.renderFile [as engine] (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:461:10)
    at View.render (/home/ubuntu/OctoFarm/node_modules/express/lib/view.js:135:8)
    at tryRender (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:640:10)
    at Function.render (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/home/ubuntu/OctoFarm/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/home/ubuntu/OctoFarm/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at /home/ubuntu/OctoFarm/routes/index.js:117:7
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)`

tvi675, 2020-04-27:
Just to add to this, since this happened it also appears to have stopped updating print statistics on the dashboard as well.

NotExpectedYet, 2020-04-28:
Cheers for the details report. 

It shouldn't be that reliant on that and I'll take a look

tvi675, 2020-04-29:
Thanks for that. Is there a way to clear the history so I can get this working again?

NotExpectedYet, 2020-04-29:
You'd have to remove the history section of your database until I can push a fix. You can nuke it you'll just loose your history you've already collected.

NotExpectedYet, 2020-05-05:
I can't reproduce this, could you try the latest development version and check if it persists please?

tvi675, 2020-05-05:
How do I update to the current dev version please? Sorry.. idiot's guide needed..

NotExpectedYet, 2020-05-05:
How did you install? 

Just to confirm this is what you did. 
I had filaments selected on a few printers which got logged into history. 

I then removed those filaments from the filament list. 

Tried to access history again.

tvi675, 2020-05-05:
git clone https://github.com/NotExpectedYet/OctoFarm.git on Ubuntu server.

Yes, that's how it happened, but was only one roll of filament on one printer. From memory I don't think filament was selected on the other printers.

NotExpectedYet, 2020-05-05:
That's weird as history saves its own reference. 
 
Woop you git cloned. 

cd into the OctoFarm folder. 

`git checkout development` 

`git pull`

`npm install`

`npm start`

btw this will start with pm2 script now. To kill it you will have to run `pm2 kill inside the OctoFarm folder.

tvi675, 2020-05-06:
Updated to the dev version as per the above, but now the web interface won't load. NPM/PM2 is running, trying to connect on port 4000 is there anything that I'm missing? Has the port changed?

NotExpectedYet, 2020-05-06:
No it hasn't. 

What's `pm2 logs` say?

tvi675, 2020-05-06:
ubuntu@ubuntu:~/OctoFarm$ pm2 logs
[TAILING] Tailing last 15 lines for [all] processes (change the value with --lin                                                                                                                                                             es option)
/home/ubuntu/.pm2/pm2.log last 15 lines:
PM2        | 2020-05-06T07:32:38: PM2 log: PM2 version          : 4.4.0
PM2        | 2020-05-06T07:32:38: PM2 log: Node.js version      : 14.0.0
PM2        | 2020-05-06T07:32:38: PM2 log: Current arch         : arm64
PM2        | 2020-05-06T07:32:38: PM2 log: PM2 home             : /home/ubuntu/.                                                                                                                                                             pm2
PM2        | 2020-05-06T07:32:38: PM2 log: PM2 PID file         : /home/ubuntu/.                                                                                                                                                             pm2/pm2.pid
PM2        | 2020-05-06T07:32:38: PM2 log: RPC socket file      : /home/ubuntu/.                                                                                                                                                             pm2/rpc.sock
PM2        | 2020-05-06T07:32:38: PM2 log: BUS socket file      : /home/ubuntu/.                                                                                                                                                             pm2/pub.sock
PM2        | 2020-05-06T07:32:38: PM2 log: Application log path : /home/ubuntu/.                                                                                                                                                             pm2/logs
PM2        | 2020-05-06T07:32:38: PM2 log: Worker Interval      : 30000
PM2        | 2020-05-06T07:32:38: PM2 log: Process dump file    : /home/ubuntu/.                                                                                                                                                             pm2/dump.pm2
PM2        | 2020-05-06T07:32:38: PM2 log: Concurrent actions   : 2
PM2        | 2020-05-06T07:32:38: PM2 log: SIGTERM timeout      : 1600
PM2        | 2020-05-06T07:32:38: PM2 log: =====================================                                                                                                                                                             ==========================================
PM2        | 2020-05-06T07:32:38: PM2 log: App [OctoFarm:0] starting in -fork mo                                                                                                                                                             de-
PM2        | 2020-05-06T07:32:38: PM2 log: App [OctoFarm:0] online

/home/ubuntu/.pm2/logs/OctoFarm-error.log last 15 lines:
0|OctoFarm | (node:2835339) Warning: Accessing non-existent property 'count' of                                                                                                                                                              module exports inside circular dependency
0|OctoFarm | (Use `node --trace-warnings ...` to show where the warning was crea                                                                                                                                                             ted)
0|OctoFarm | (node:2835339) Warning: Accessing non-existent property 'findOne' o                                                                                                                                                             f module exports inside circular dependency
0|OctoFarm | (node:2835339) Warning: Accessing non-existent property 'remove' of                                                                                                                                                              module exports inside circular dependency
0|OctoFarm | (node:2835339) Warning: Accessing non-existent property 'updateOne'                                                                                                                                                              of module exports inside circular dependency
0|OctoFarm | (node:1453) Warning: Accessing non-existent property 'count' of mod                                                                                                                                                             ule exports inside circular dependency
0|OctoFarm | (Use `node --trace-warnings ...` to show where the warning was crea                                                                                                                                                             ted)
0|OctoFarm | (node:1453) Warning: Accessing non-existent property 'findOne' of m                                                                                                                                                             odule exports inside circular dependency
0|OctoFarm | (node:1453) Warning: Accessing non-existent property 'remove' of mo                                                                                                                                                             dule exports inside circular dependency
[... cut ...]
0|OctoFarm | Wed, 06 May 2020 07:27:51 GMT | ERROR | OctoFarm-Server.log | Mongo                                                                                                                                                             oseServerSelectionError: connection timed out |
0|OctoFarm | db: mongodb://192.168.1.5:27017/octofarm
0|OctoFarm | Wed, 06 May 2020 07:33:10 GMT | ERROR | OctoFarm-Server.log | Mongo                                                                                                                                                             oseServerSelectionError: connection timed out |

NotExpectedYet, 2020-05-06:
It's timing out connecting to your mongodb server

NotExpectedYet, 2020-05-06:
It's overwritten your db.js file to mine, Update config/db.js 

Shouldn't have done that I'll take a look at gitignore

tvi675, 2020-05-06:
Ah, that's interesting. When I initially set it up I changed the server IP to localhost:27017, so that seems to have been overwritten with the upgrade

Crowlord, 2020-05-06:
gonna chip in as I just set it up. The db.js file has changed. 

I had to change my mongo DB back to localhost:4000 its currently set to

Crowlord, 2020-05-06:
PS . the new dev version looks GREAT

tvi675, 2020-05-06:
Upgrade issue resolved by changing db.js.
History issue not resolved, still showing

TypeError: /home/ubuntu/OctoFarm/views/history.ejs:56
    54|             <% if(typeof print.printHistory.filamentSelection != 'undefined' &&
    55|             print.printHistory.filamentSelection != "None chosen..."){%>
 >> 56|               <% let
    57|             filament = print.printHistory.filamentSelection.roll %> <%=
    58|             filament.name %> [<%= filament.colour %> / <%= filament.type[1] %>]
    59|             <%}else{%> None selected... <% } %>

Cannot read property 'roll' of null
    at eval (/home/ubuntu/OctoFarm/views/history.ejs:58:61)
    at Array.forEach (<anonymous>)
    at eval (/home/ubuntu/OctoFarm/views/history.ejs:13:16)
    at history (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:662:17)
    at tryHandleCache (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:255:36)
    at View.exports.renderFile [as engine] (/home/ubuntu/OctoFarm/node_modules/ejs/lib/ejs.js:461:10)
    at View.render (/home/ubuntu/OctoFarm/node_modules/express/lib/view.js:135:8)
    at tryRender (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:640:10)
    at Function.render (/home/ubuntu/OctoFarm/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/home/ubuntu/OctoFarm/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/home/ubuntu/OctoFarm/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at /home/ubuntu/OctoFarm/routes/index.js:126:9
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)

tvi675, 2020-05-06:
Happy to reset that part of the database if you can point me in the right direction

NotExpectedYet, 2020-05-06:
Strange is that one, dunno how it's happened at all. 

I'll see if there's another check I can put in but the filament should be there and logged. 

On ubuntu command line I think it's: 

`mongo`

`use <octofarm database name>`

`show collections`

`db.histories.drop()`

tvi675, 2020-05-06:
Reset the database, now all working again. Very strange indeed.

NotExpectedYet, 2020-05-06:
Sweet, I'll run a few more tests something must have borked up. :)

NotExpectedYet, 2020-05-13:
I just nuked the old filament database and it still loads -_- no idea what happened here man but I'm going to close for now. Please re-open if it happens again or we can get some specific steps to take to reproduce.

closed 2020-05-13 by NotExpectedYet