OctoFarm/OctoFarm #43: Raspian OS on a Raspberry Pi 4 (does not work)

cskozlowski, 2020-04-28:
Do not install OctoFarm on a RPI4 running under Raspian,  install Ubuntu OS (64 bit) and this will work.  There is a MongoDB version issue (2.4.xx) under under Raspian and this will not function properly with OctoFarm.  A later version of MongoDB (3.x.x) is used with the Ubuntu release and this works fine with OctoFarm.

NotExpectedYet, 2020-04-28:
Please read the README as it states which version of MongoDB is supported. Closing because this isn't an issue with the correct version.

closed 2020-04-28 by NotExpectedYet

cskozlowski, 2020-04-28:
I just found that reference(MongoDB 3.6+).  Might consider stating explicitly that this version of MongoDB is not available on a RPI running Raspian OS .  I bet there are other folks out there who will try what I did since Octoprint running under Raspian OS which will not work.

NotExpectedYet, 2020-04-28:
It already states it:
Note: Raspberry Pi's Raspbian OS doesn't officially support running MongoDB yet (MongoDB requires a 64bit kernel, whereas Raspbian's is 32bit), so in that case, you'll need an external database running on some other machine or VM.

cskozlowski, 2020-04-28:
Yes it does, however, the wording could be changed from 'doesn't officially support running' to 'doesn't work with'.  When I read that a piece of software 'doesn't officially support', that is a warning that I might/will have issues.  MongoDB will install under Raspbian  and I went into about a 2 hour time hole in order to figure out that Raspbian is a non-starter.  I first had to figure out which version of MongoDB got installed (v2.4.xxx) and then started hunting for ways to upgrade to a later version of MongoDB under .  I am new to MongoDB so that is part of the problem as well.

NotExpectedYet, 2020-04-28:
To me those are the same thing, with the added clarification of "MongoDb requires a 64Bit Kernel". 

Anyway, thanks! Glad you like it.