OctoFarm/OctoFarm #45: Recommended Automatic start of npm (under linux)?

cskozlowski, 2020-04-28:
Is there a recommended method for autostarting npm after a bootup is complete under linux.  I have attempted to do this using init.d and have had no success.

NotExpectedYet, 2020-04-28:
a lot use an npm package called pm2. 

You can install with `npm install pm2 -g`

Then run the following. 

`sudo pm2 startup`

 `pm2 start app.js --name OctoFarm`

`pm2 save`

https://pm2.keymetrics.io/docs/usage/startup/

There's more information about it here for you, let me know if you get stuck. 

1.1.5 is coming with pm2 as a dependency and it runs using that, I will hopefully be able to enable the startup commands at boot so it's all automatic but I've not looked much into it yet.

gmccauley, 2020-04-28:
I was using a systemd service file with success using Ubuntu.  As James states, the newer version is coming with pm2.  With pm2, you can run `pm2 startup` and it will give detect your init system and output a command to run to generate the startup files.  I just did it and it seems to work pretty good.

FULL DISCLOSURE:  I'd never even heard of pm2 before today so I had to do a little research and this is what I found and what I did and it appears to work.

Here's what happens when I run `pm2 startup`
```
ubuntu@octofarm:~$ OctoFarm/node_modules/pm2/bin/pm2 startup
[PM2] Init System found: systemd
ubuntu
[PM2] To setup the Startup Script, copy/paste the following command:
sudo env PATH=$PATH:/usr/bin /home/ubuntu/OctoFarm/node_modules/pm2/bin/pm2 startup systemd -u ubuntu --hp /home/ubuntu
```

And then after I run the command it told me:
```
ubuntu@octofarm:~$ sudo env PATH=$PATH:/usr/bin /home/ubuntu/OctoFarm/node_modules/pm2/bin/pm2 startup systemd -u ubuntu --hp /home/ubuntu
[PM2] Init System found: systemd
Platform systemd
Template
[Unit]
Description=PM2 process manager
Documentation=https://pm2.keymetrics.io/
After=network.target

[Service]
Type=forking
User=ubuntu
LimitNOFILE=infinity
LimitNPROC=infinity
LimitCORE=infinity
Environment=PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/usr/bin:/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Environment=PM2_HOME=/home/ubuntu/.pm2
PIDFile=/home/ubuntu/.pm2/pm2.pid
Restart=on-failure

ExecStart=/home/ubuntu/OctoFarm/node_modules/pm2/bin/pm2 resurrect
ExecReload=/home/ubuntu/OctoFarm/node_modules/pm2/bin/pm2 reload all
ExecStop=/home/ubuntu/OctoFarm/node_modules/pm2/bin/pm2 kill

[Install]
WantedBy=multi-user.target

Target path
/etc/systemd/system/pm2-ubuntu.service
Command list
[ 'systemctl enable pm2-ubuntu' ]
[PM2] Writing init configuration in /etc/systemd/system/pm2-ubuntu.service
[PM2] Making script booting at startup...
[PM2] [-] Executing: systemctl enable pm2-ubuntu...
Created symlink /etc/systemd/system/multi-user.target.wants/pm2-ubuntu.service → /etc/systemd/system/pm2-ubuntu.service.
[PM2] [v] Command successfully executed.
+---------------------------------------+
[PM2] Freeze a process list on reboot via:
$ pm2 save

[PM2] Remove init script via:
$ pm2 unstartup systemd
```

cskozlowski, 2020-04-28:
Thanks for the quick response to both of you.  Will try this later today.  I love/hate Linux systems - about 10 different ways to do everything.  I started with BSD Unix back in the late 1980s.

NotExpectedYet, 2020-04-29:
No trouble, I'll leave this open for now to remind me to add some notes to the README/Wiki

NotExpectedYet, 2020-06-04:
Installation documnets added for doing this cross platform

closed 2020-06-04 by NotExpectedYet