OctoFarm/OctoFarm #47: Unable to add ?

Totalgeek9224, 2020-05-13:


Totalgeek9224, 2020-05-13:
Whenever i attempt to add or remove a printer, sin the first one i added, The Octofarm web instance just loads, until it times out. I have tried uninstalling and reinstalling Octofarm, yet the initial printer added still remains, and the issue persists

NotExpectedYet, 2020-05-13:
Have you tried clearing your database and then starting afresh? 

Deleting and Adding printers isn't the best in that version.

Totalgeek9224, 2020-05-13:
V1.14. Pulled from Git yesterday. I reinstalled today using thetinkerdads's docker method, and while now I can seemingly add printers, what printer i HAVE added are not showing as active in the Octofarm platform. I have ensured that CORS is enabled on the pi in question, and it has been restarted multiple times

NotExpectedYet, 2020-05-13:
If you mean 1.1.4.6 then you want to go back to that version as tinkerdads docker is on 1.1.4. Which is why your printers won't connect as I released a bug fix for that version. 

Just change the URL endpoint in config/db.js to something else other then OctoFarm and it will set you up fresh.

Totalgeek9224, 2020-05-13:
As it stands, the docker run line is
 RUN sed -i "s/192.168.1.5:27017\/octofarm/$MONGOUSER:$MONGOPASS@$MONGOHOST:27017\/admin/g" ./config/db.jsRUN sed -i "s/192.168.1.5:27017\/octofarm/$MONGOUSER:$MONGOPASS@$MONGOHOST:27017\/admin/g" ./config/db.js

Seeing as the endpoint isnt octofarm, whats the best way to go about this?

NotExpectedYet, 2020-05-13:
Is that from tinkerdads docker? I'm advising you not to use that if it is. 

How did you set it up originally before tinkerdads?

Totalgeek9224, 2020-05-13:
Okay, so, originally had set it up using Chris' basement's video, which lead to the initial issue. Then, uninstalled it (i think), and then went with the tinkerdad method. This is all on windows,

NotExpectedYet, 2020-05-13:
right so the reason why uninstalling it did nothing is because it was still connecting to your old database with the problem printers on it. 

All you have to do is re-edit config/db.js and instead of /OctoFarm at the end of your mongo URL you just change it to /octofarm2 or whatever you want really. Then re-start OctoFarm. Just make sure tinkerdads docker is killed as it will lock port 4000. 

Then re-add your printers and you should be fine. Any issues give me a shout.

NotExpectedYet, 2020-05-20:
Closing this as I think the original question has been answered. Please log a new ticket if not the case.

closed 2020-05-20 by NotExpectedYet