OctoFarm/OctoFarm #51: OctoFarm does not establish a connection

3DFans, 2020-05-17:
The installation works so far without problems and there are no problems. If I now enter the IP followed by 4000 in the browser, no OctoFarm page appears.

NotExpectedYet, 2020-05-17:
Usually that's an issue with your database. 

How have you set it up?

3DFans, 2020-05-17:
Hey, wow very quick answer. Top
I set up MangoDB and then edited the IP on localhost. After that I did nothing more and followed the instructions exactly.

NotExpectedYet, 2020-05-17:
:+1: 

On which OS have you installed? 

There should be a production.log and a productionError.log or some command line output, do you have any of those? 

There's a recent bug I've just been notified about with nodeJS version 14 too, could you check your node version with `node -v` as well please.

3DFans, 2020-05-17:
The two lines appear at the end after I type nps start. Windows 10 is used and the node version is 13.14.0

NotExpectedYet, 2020-05-17:
Look in your OctoFarm/logs folder there should be two files in there.

3DFans, 2020-05-17:
`
db: mongodb://localhost:27017/octofarm
MongooseError [MongooseServerSelectionError]: Server at localhost:27017 reports maximum wire versio$
    at new MongooseServerSelectionError (/home/pi/OctoFarm/node_modules/mongoose/lib/error/serverSe$
    at NativeConnection.Connection.openUri (/home/pi/OctoFarm/node_modules/mongoose/lib/connection.$
    at Mongoose.connect (/home/pi/OctoFarm/node_modules/mongoose/lib/index.js:333:15)
    at Object.<anonymous> (/home/pi/OctoFarm/app.js:70:4)
    at Module._compile (internal/modules/cjs/loader.js:1118:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1138:10)
    at Module.load (internal/modules/cjs/loader.js:982:32)
    at Function.Module._load (internal/modules/cjs/loader.js:875:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:71:12)
    at internal/main/run_main_module.js:17:47 {
  reason: TopologyDescription {
    type: 'Single',
    setName: null,
    maxSetVersion: null,
    maxElectionId: null,
    servers: Map(1) { 'localhost:27017' => [ServerDescription] },
    stale: false,
    compatible: true,
    compatibilityError: null,
    logicalSessionTimeoutMinutes: null,
    heartbeatFrequencyMS: 10000,
    localThresholdMS: 15,
    commonWireVersion: null
  },
  [Symbol(mongoErrorContextSymbol)]: {}
}
`
productionError ist clear

NotExpectedYet, 2020-05-17:
hmm not sure on that one. 

what version if your database?

3DFans, 2020-05-17:
4.2.5

NotExpectedYet, 2020-05-17:
Have you tried with 127.0.0.1 instead of localhost? 

and the database is certainly running? 

I'm at a loss with that one I'll have to investigate a bit more.

3DFans, 2020-05-17:
It doesn't work with IP either. The following is stated and therefore I assume that it will run correctly.

> octofarm@1.1.4 start / home / pi / OctoFarm
> node app.js> logs / production.log 2> logs / productionError.log

How do I start the database directly?

NotExpectedYet, 2020-05-17:
no that just means my app has started. The logs indicate it can't connect to your mongodb instance. 

I've personally never installed it on windows so unless it's running as a service and you can logon to the mongodb command line then I'll have to do some research to help.

3DFans, 2020-05-17:
Oh, no, I installed it on the Raspberry, sorry. : D

NotExpectedYet, 2020-05-17:
🤣

right well rasbian doesn't support mongodb 64 bit. You'll have to install Ubuntu 18.04 64-bit version if you want to run it on a Pi. 

Closing as I believe the original issue has been solved.

3DFans, 2020-05-17:
:D
Ubuntu 32bit or 64bit ?

NotExpectedYet, 2020-05-17:
You need 64-bit for the MongoDB :)

3DFans, 2020-05-17:
ok thanks for your help =)

closed 2020-05-17 by 3DFans