OctoFarm/OctoFarm #53: Bug Fix Release 3 cannot delete a printer if its Raspberry Pi is offline or in a 'shutdown' state.

iandouglas, 2020-05-20:
I have 6 printers configured and OctoFarm has been working great. I tried to delete a printer tonight and the UI says it deleted the printer successfully and to click the Save button, then I get a spinning animation while the server restarts, but the animaion never goes away.

Doing a `tail -f logs/productionError.log` in another terminal gives the following error:
```txt
(node:3027925) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'close' of undefined
    at Function.reset (/home/id/src/public/octofarm/runners/state.js:452:41)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)
    at async /home/id/src/public/octofarm/routes/printers.js:71:17
(node:3027925) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
```

`state.js` line 452 is inside a `static async reset()` method where it looks like maybe it's trying to close the connection to all printers in sequence. I not currently able to get into a debugger to see what else is happening.

While I'm trying to do this, 4 of my printers are printing, one is in a "closed" mode because that OctoPrint instance is only utilizing the camera while the printer prints from SD card. The 6th printer's raspberry pi is already unplugged and seems stuck in a "shutdown" status.

Refreshing the page shows that all printers are still configured including the one I'm trying to delete that the UI said had already been deleted. Perhaps the UI should say "printer is queued to be removed on server restart" instead of the current message saying the printer IS removed?

If I power on that Raspberry Pi and refresh OctoFarm until that unit is connected to the farm, I can successfully remove it without any errors.

We should be able to delete a printer from the farm whether the Raspberry Pi is online and working correctly or not.

NotExpectedYet, 2020-05-20:
Thanks for the detailed report. 

Yeah there's a lot not accounted for in that current release. It's a lot more stable now in the later 1.1.5 which is currently un-released. You are welcome to try it on the development branch but there is no support given until it's fully released. 

Just out of curiosity why did you want to remove the printer from OctoFarm? 

If the UI is correctly responding now to the deleted printer then you should be fine going forward. The only other option currently would be to delete the specific printer out of the database if that's needed and restart OctoFarm and it will start afresh.

iandouglas, 2020-05-20:
I was trying to remove a printer because its Raspberry Pi was failing, locking up, etc. I replaced the Raspberry Pi with a new unit but had to rummage through some storage bins to find another older raspberry pi to take the SD card and boot far enough along that I could get OctoFarm to remove it. Reasons aside, it shouldn't matter why -- if a user wants to delete something from the farm it should have the ability to do so even if the associated OctoPrint instance is unavailble.

I'm also curious why the OctoFarm software has to restart when removing a printer, when I don't recall that a restart was needed when adding a printer? (just asking for curiosity's sake)

I haven't looked into the database schema in MongoDB at all to know how deleting a printer would impact other things, which is why I'd prefer to use the UI to do so.

NotExpectedYet, 2020-05-21:
Thanks for the explanation, so really you want an edit function not a remove and replace. I asked to ascertain that. 

The code base is new, and myself to coding. There are going to be bugs, simple as that really.

iandouglas, 2020-05-21:
I think having the full ability to just delete a printer is still a useful function. One of the printers I'm running right now belongs to my school and I do need to return it at some point. And the UI looks like we *can* remove a printer, provided that its respective Raspberry Pi is working. :)

NotExpectedYet, 2020-05-21:
Aye complete oversight on my part. I resolved a lot of this for 1.1.5 but it's still in heavy development currently. It's stable just some half finished features as it is and requires some testing which me and a few are going through at the moment. I'll double check this is actually possible with the latest code as well just in-case the eventuality ever arises again.

NotExpectedYet, 2020-06-04:
Fixed in recent 1.1.5 push

closed 2020-06-04 by NotExpectedYet