OctoFarm/OctoFarm #55: "Print Harvested" not updating

ID4600, 2020-05-28:
After print is finished, when I press the "Print Harvested?" button nothing happens, it stays in the same state.

Running: 1.1.4-6-bugfix
OS: Win10 Pro

Happens on all available printers. Fresh install did not resolve the problem.

NotExpectedYet, 2020-05-28:
Have you enabled CORS and restarted?

NotExpectedYet, 2020-05-28:
On OctoPrint

ID4600, 2020-05-28:
Okay that was a stupid thing to mis.....

Only thing that I can't fix is the syncing with the printers files, some printers do show the files, others simply don't even after uploading through octofarm, they disappear after a bit.

NotExpectedYet, 2020-05-28:
Easy fix though. 

How many files are on them? It doesn't wait long enough if they aren't there. 

The disappearing is a new one though I think.

ID4600, 2020-05-28:
There should be around 80 files here, maybe that's the problem? 
There are 10 folders and 57 files in the root. 
[image]

This one was uploaded through the interface of octofarm: 

[image]

NotExpectedYet, 2020-05-28:
Yeah looks like the timeout isn't long enough. You can edit serverConfig/server.js in the OctoFarm folder to give it a little more time. There's two settings to change in there you can try to increase for API timeout. Should give it enough time to grab those files then.

NotExpectedYet, 2020-05-28:
I'm gonna label this a bug and leave it here actually. It needs looking into more than just increasing the timeout.

ID4600, 2020-05-28:
I've set serverConfig/timeout.js line 3 to "apiTimeout: 8000," seems to solve it for now!

NotExpectedYet, 2020-05-28:
Fixed this on the development branch. Will be released soonish I hope.

commit 250a1800 (2020-05-28): File Manager timeout fix

#Fixes #55
closed 2020-06-04 by NotExpectedYet