OctoFarm/OctoFarm #57: History page not displayed after remove a filament

facaf, 2020-05-31:
After deleting a roll of filament, the history page shows error.
I am using a 18.04 linux server (console only) with octoprint 1.4.0 with 4 instances.
If you can help me, I will be very grateful.

TypeError: /home/facu/OctoFarm/views/history.ejs:57
    55|             <% if(typeof print.printHistory.filamentSelection != 'undefined' &&
    56|             print.printHistory.filamentSelection != "None chosen..."){%>
 >> 57|               <% let
    58|             filament = print.printHistory.filamentSelection.roll %> <%=
    59|             filament.name %> [<%= filament.colour %> / <%= filament.type[1] %>]
    60|             <%}else{%> None selected... <% } %>

Cannot read property 'roll' of null
    at eval (/home/facu/OctoFarm/views/history.ejs:60:61)
    at Array.forEach (<anonymous>)
    at eval (/home/facu/OctoFarm/views/history.ejs:13:16)
    at history (/home/facu/OctoFarm/node_modules/ejs/lib/ejs.js:662:17)
    at tryHandleCache (/home/facu/OctoFarm/node_modules/ejs/lib/ejs.js:255:36)
    at View.exports.renderFile [as engine] (/home/facu/OctoFarm/node_modules/ejs/lib/ejs.js:461:10)
    at View.render (/home/facu/OctoFarm/node_modules/express/lib/view.js:135:8)
    at tryRender (/home/facu/OctoFarm/node_modules/express/lib/application.js:640:10)
    at Function.render (/home/facu/OctoFarm/node_modules/express/lib/application.js:592:3)
    at ServerResponse.render (/home/facu/OctoFarm/node_modules/express/lib/response.js:1012:7)
    at ServerResponse.res.render (/home/facu/OctoFarm/node_modules/express-ejs-layouts/lib/express-layouts.js:77:18)
    at /home/facu/OctoFarm/routes/index.js:117:7
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (internal/process/task_queues.js:97:5)

NotExpectedYet, 2020-06-01:
Currently you can either delete the history table in the database or switch over to the development branch and see if it helps the issue. 

Seems there is still a link somewhere even though I couldn't reproduce this by just capturing history and I thought there was already a ticket with another but I think it got closed. 

If there's some specific steps you took would be good to know :)

facaf, 2020-06-01:
Hello! Thanks for answering.
Any specific path and file to delete?
The steps taken to generate the problem were:
1) Define 2 filaments that I usually use (with all the fields)
2) print without choosing filament in printer-filament
3) in the course of printing, change the used filament.

From that moment it was impossible for me to restore history.

From console I tried to find the archive, but I couldn't find it.

NotExpectedYet, 2020-06-01:
Thanks! I will make sure the latest version doesn't have this bug. 

For deletion you will need to log onto the MongoDB database and delete the table called History. 

`mongo` - Will get you onto the MongoDB shell
`db.histories.drop()` - Should delete the history table.

facaf, 2020-06-01:
111/5000
Thank you! I was able to delete the history from the database.
I close the issue, and if notice appears again.
Regards.

closed 2020-06-01 by facaf