OctoFarm/OctoFarm #58: Add Printer by URL Doesn't Connect

zdesignstudio, 2020-06-04:
When addign a printer via URL (feature request added in 1.1.5), OctoFarm will not connect to OctoPrint. Status shows as Shutdown/Offline.
[image]

NotExpectedYet, 2020-06-05:
I would gather that's the trailing slash you've got in the URL?

zdesignstudio, 2020-06-05:
Man you are good.

zdesignstudio, 2020-06-05:
Jumped the gun.

Removed the trailing slash and it showed as online but then went to Shutdown and keeps going between Shutdown and Online.

NotExpectedYet, 2020-06-05:
are you running this through a reverse proxy? I'm not sure how the websockets will be handled but you'd probably need to forward them.

NotExpectedYet, 2020-06-05:
I've not specifically tested this feature myself but the websockets will connect to ws:// but if you let me know a bit more about your setup I can try and replicate here and make some recommendations :)

zdesignstudio, 2020-06-05:
Using the built in haproxy to redirect the path name to ip:port. Here is one of my haproxy configs on one of my PIs.

 

 

 

 

global

        maxconn 4096

        user haproxy

        group haproxy

        log 127.0.0.1 local1 debug

 

defaults

        log     global

        mode    http

        option  httplog

        option  dontlognull

        retries 3

        option redispatch

        option http-server-close

        option forwardfor

        maxconn 2000

        timeout connect 5s

        timeout client  15min

        timeout server  15min

 

frontend public

        bind *:80

        bind 0.0.0.0:443 ssl crt /etc/ssl/snakeoil.pem

        option forwardfor except 127.0.0.1

        use_backend webcam if { path_beg /webcam/ }

        errorfile 503 /etc/haproxy/errors/503-no-octoprint.http

        use_backend prusa if { path_beg /prusa/ }

        use_backend cr10 if { path_beg /cr10/ }

        use_backend cr10s if { path_beg /cr10s/ }

        use_backend wanhao if { path_beg /wanhao/ }

        default_backend webcam

 

backend prusa

        reqrep ^([^\ :]*)\ /prusa/(.*) \1\ /\2

        option forwardfor

        server octoprint1 127.0.0.1:5000

        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }

        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }

        reqadd X-Script-Name:\ /prusa

 

backend cr10

        reqrep ^([^\ :]*)\ /cr10/(.*) \1\ /\2

        option forwardfor

        server octoprint2 127.0.0.1:5001

        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }

        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }

        reqadd X-Script-Name:\ /cr10

 

backend cr10s

        reqrep ^([^\ :]*)\ /cr10s/(.*) \1\ /\2

        option forwardfor

        server octoprint3 127.0.0.1:5002

        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }

        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }

        reqadd X-Script-Name:\ /cr10s

 

backend wanhao

        reqrep ^([^\ :]*)\ /wanhao/(.*) \1\ /\2

        option forwardfor

        server octoprint4 127.0.0.1:5003

        acl needs_scheme req.hdr_cnt(X-Scheme) eq 0

        reqadd X-Scheme:\ https if needs_scheme { ssl_fc }

        reqadd X-Scheme:\ http if needs_scheme !{ ssl_fc }

        reqadd X-Script-Name:\ /wanhao

 

backend webcam

        reqrep ^([^\ :]*)\ /webcam/(.*)     \1\ /\2

        server webcam1  127.0.0.1:8080

 

userlist L1

        group G1

        user admin insecure-password Password123 groups G1

NotExpectedYet, 2020-06-05:
Hmmm could you pass me the OctoFarm-State.log file as well please? 

Just double check where it's failing the connection.

NotExpectedYet, 2020-06-05:
Where does the password get asked for btw? I have no capabilities for a passworded instance currently. Not sure if it would affect anything just trying to remove all the unknowns.

zdesignstudio, 2020-06-05:
So I pulled the logs and it was still showing the trailing / and OF was adding an additional / for the remainder of the ws url.

Fri, 05 Jun 2020 03:48:15 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"1009: ws://10.10.1.31/cr10s**//**sockjs/websocket" | 
Fri, 05 Jun 2020 03:48:15 GMT | INFO | OctoFarm-State.log | WebSocketClient: Closed | data:{"0":1006} 

I restarted OctoFarm and it looks to be ok now. Not sure why it kept the / after I removed it though and required a restart.

The above issue is repeatable also... Add URL with a trailing / save. Remove the / Save, logs still show the /. Restart the service and its resolved.

NotExpectedYet, 2020-06-05:
Somethings not been cleared out with that trailing slash then, I'll double check the edit function is working properly thanks! 

I'll leave this issue open until I can do some investigation into why as the database is been updated but it seems to not be properly re-scanning with the updated information.

zdesignstudio, 2020-06-05:
Just a little more insight on the issue.
I added the / back in and saved and it took without issue causing the state to remain in Shutdown status. I removed the / and the issue reoccurred with it bouncing back between Online and Shutdown. Restart of instance fixed it.
Thanks.

NotExpectedYet, 2020-06-05:
Yeah it's not clearing something. 
I just tried switching from 5001 to 5002 on my own instances here and it does the bouce between the offline/online states. Both my instances have the same API key. A re-scan doesn't even do anything either. 
Thanks for the reports!

zdesignstudio, 2020-06-05:
Anytime

NotExpectedYet, 2020-06-05:
Fixed, I'll get master updated in due course :)

commit f355a531 (2020-06-05): Fixed #58
Printer didn't close properly on edit
closed 2020-06-05 by NotExpectedYet