OctoFarm/OctoFarm #63: Upgrade process

jfkerekes, 2020-06-05:
What is the upgrade process for a Windows 10 desktop install of Octofarm? I would like to upgrade to the latest release.

NotExpectedYet, 2020-06-05:
How did you install it? 

git clone

or downloading the .zip?

jfkerekes, 2020-06-05:
Downloaded the ZIP and, just to kluge it and get it working, I stopped the PM2 instance of 1.1.4 and saved the PM2 settings. Then I archived the 1.1.4  install folder and then did NPM install for the 1.1.5 update which gotme p and running. Ran into a couple of gotcha's, however, and I am not sure if it is because you made changes to the Mongodb Database structure or if it is because of the way I updated... It pulled in all of my printer, however, while it saw that I had files on each of my octoprint servers, only the first printer that I set up when I did the 1.1.4 install would actually display any of those files. I have removed and re-added 1 printer and now the files on that Octoprint server are showing in Octofarm. The other thing that I noticed... I had set up several different filaments in 1.1.4, which are now not showing in 1.1.5 which I am assuming is because of the new filament management system.

garyriet, 2020-06-05:
Is there a write up on how to upgrade?

NotExpectedYet, 2020-06-06:
@garyriet there's no write up for windows, but every installation instruction in the wiki should have update path at the end. 

Windows is pretty similar though, download the new and re-run the npm install, npm start commands after making sure there is a line in the database. 

@jfkerekes Yeah a lot changed in the database this version from the old. It's mentioned in all the release notes. The old filament one. 

It looks like the original question has been answer so I'm going to close. If the files are still an issue please open a new ticket.

closed 2020-06-06 by NotExpectedYet