OctoFarm/OctoFarm #65: Host State Keeps Disconnecting

Crane49, 2020-06-07:
I am currently running octofram 1.1.5.5. When I add a printer it connects then imediately disconects from octofarm. When I resync, I get a connection but it immediately disconnects. I have 3 printers with all printers and OctoFarm hard wired on my network. I was previously runing OcotFarm 1.1.4.6 without issue. I run OctoFarm on RPi 4b 4gb headless on Ubuntu 20.04 server. I will try to change the intervals times to see if that is an issue??
[image]

Thanks,
Mike

NotExpectedYet, 2020-06-07:
http://172.31.156.20/
Should be:
http://172.31.156.20

Crane49, 2020-06-07:
Thank you. Everything is working!

closed 2020-06-07 by Crane49

NotExpectedYet, 2020-06-07:
No trouble! 

Out of curiosity did you input those yourself or did it happen as part of the conversion? OctoFarm should detect the old IP's and re-write them correctly so I'll take a look if it isn't.

Crane49, 2020-06-07:
I did do the input manaully.

NotExpectedYet, 2020-06-07:
Sound no trouble, thanks for the reply :) 

Happy printing aha