OctoFarm/OctoFarm #67: Websocket terminated. Ping/ Pong check failed.

Crowlord, 2020-06-11:
My printers regularly get reported as offline. Octofarm reports the websocket as terminated as the ping/pong check failed. Adjusting timeouts seems to make no difference.

NotExpectedYet, 2020-06-11:
What version are you on please?

NotExpectedYet, 2020-06-11:
Could you also attach the latest state log where this is happening as well please?

Crowlord, 2020-06-11:
This is on 1.1.5.5 running on windows 10. 

[log copy.txt](https://github.com/NotExpectedYet/OctoFarm/files/4764812/log.copy.txt)
[image]

jrmartinezr, 2020-06-11:
Hi, here the same issue.  

OctoFarm Version 1.1.5.5
Running on Ubuntu 20.04 x64
Raspberry PI 3B+
nodejs v13.14.0
mongo v3.6.8
npm v6.14.4
OctoPrint 1.4.0 on Raspberry PI 3B+

[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/4765011/OctoFarm-State.log)

[image]

Greetings, take care.

jfkerekes, 2020-06-11:
Hi, 

Octofarm 1.1.5.4
Running on Windows 10
Mix of RPI 3s and 4s
Octoprint 1.4.0

Having same issue... printers will sporadically show as offline and forcing a re-sync corrects it. Changing timeouts does not seem to make a difference.

jfkerekes, 2020-06-11:
Just upgraded to 1.1.5.5 and still having same issue

NotExpectedYet, 2020-06-11:
I have just pushed a change to development if anyone wants to see if it's fixed it.

jfkerekes, 2020-06-11:
I would be happy to test

Crowlord, 2020-06-11:
I will jump on the dev branch and let you know

jfkerekes, 2020-06-11:
Installed 1.1.5.6 and started server and it pulled all 4 of my printers status immediately and showed as online. Can say for sure yet but so far it looks better.

jfkerekes, 2020-06-11:
Just noticed 2 things not working... and I understand this is a dev build, but right now when I go into a printer to control/start a print, T0 temp registers and shows set and actual temps, Bed temp, however, does not show set or actual temp. I also noticed that even though I have filament assigned to a printer, it shows that filament assigned in the list view but when I open a particular printer to start a print, the filament drop down does not show the assigned filament and is not populated with any of the filaments in the database. So far, however, I have not lost connection to any of the printers sporadically.

NotExpectedYet, 2020-06-12:
There's major development on this version it was just to test the connections. Good to know. Id go back to the previous version until I can finish the changes for this version. 

To fix that version it's literally one single line change if you want? Then your temps and filament will work again. 

Runner/stage.js on line 55. 
`               if (client.ws.isAlive === false) return client.ws.instance.close();`
Should become:
`               if (client.ws.isAlive === false) return client.ws.instance.terminate();`

Crowlord, 2020-06-12:
I have run overnight and both are still showing online. 

Looks like original issue is confirmed fixed in dev.

NotExpectedYet, 2020-06-12:
Sweet. there's instructions above to apply the fix to the old version as dev is currently broken with the tool0 changes that are coming to support multiple tools/chambers. When those are complete I will push all to master

commit 47ca0187 (2020-06-11): Fixes #67
commit 531d86c8 (2020-06-11): Fixed #67
commit cf492a07 (2020-06-27): Fixed #67

Stopped the ping/pong check disabling rechecking for printers
closed 2020-06-27 by NotExpectedYet