OctoFarm/OctoFarm #69: Camera View Breaks

Austin3dPrinting, 2020-06-11:
Hey there, first off great work; loving this software and the new features!

I noticed that Panel view just about always works as expected and shows the desired information and camera view. However, when viewing the camera page the streaming feed tends to not show unless I modify a section of the camera page settings and then load it back again. That tends to work every time.

Feedback happily received in case I set something up incorrectly on my linux Ubuntu Server deployment. Screen grab video attached showcasing what I experienced.
[OctoFarm Camera View Issue.zip](https://github.com/NotExpectedYet/OctoFarm/files/4766177/OctoFarm.Camera.View.Issue.zip)

NotExpectedYet, 2020-06-11:
That's a new one and one I don't specifically experience myself so I wonder if it's MAC specific. The weirdest part is when you traverse those pages everything is destroyed and recreated. 

I'll have to see if I can get a Mac VM and replicate.

NotExpectedYet, 2020-06-11:
Does reloading again fix it as that's all applying that setting would do.

Austin3dPrinting, 2020-06-11:
Oddly enough no, that was my first go to. I'm uploading another screencast showing this attempt and the outcome.
[Screen Recording 2020-06-11 at 2.44.14 PM.zip](https://github.com/NotExpectedYet/OctoFarm/files/4766687/Screen.Recording.2020-06-11.at.2.44.14.PM.zip)

NotExpectedYet, 2020-06-12:
That's weird, certainly sounds browser specific. Does the issue persist when you use a different browser like chrome?

Austin3dPrinting, 2020-06-12:
Thank you for looking into it, I did test it on Chrome and couldn't reproduce the behavior so I believe you're correct that it is a Safari issue, as is the malfunctioning double click for fullscreen feature that works elsewhere like Chrome and FireFox. Think there is anything I can do on my end to make it work with Safari?

Thank you again for this awesome software!

NotExpectedYet, 2020-06-13:
Certainly is then, thanks for double checking that!

No worries as well :) I'd suggesting just using Chrome for now as it will be something in the code Safari doesn't like so I'm not sure what you can do. 

I'll take a look at the fullscreen function certainly that could be tweaked to allow to work on safari but the camera loading seems specific to how they do things but I'll leave this ticket open so I don't forget to have a look.

NotExpectedYet, 2020-08-17:
Hello I'm going to close this issue as Safari isn't one of the supported browsers. Due to me been a one man team I'm only going to be focusing on Firefox and Chrome.

closed 2020-08-17 by NotExpectedYet