OctoFarm/OctoFarm #70: Filament Manager Plugin?

tonysurma, 2020-06-12:
Apologies if in the docs somewhere and I just didn't find it but looks like latest release integrates with a/the filament manager plugin?  I am presuming this [one](https://github.com/malnvenshorn/OctoPrint-FilamentManager)?

If so do you have experience with it.  It unfortunately looks like it hasn't been updated in over a [year plus](https://github.com/malnvenshorn/OctoPrint-FilamentManager/issues/210) and was recently [marked as abandoned](https://github.com/OctoPrint/plugins.octoprint.org/issues/471).

Just curious as I totally wanted this capability and feature and wasn't sure about investing in this plugin & database if it is either not working due to no new updates or abandoned.

Thanks again for this system, I use it everyday

NotExpectedYet, 2020-06-13:
Yeah that's the one. It still works fine for me. 

I'm going to be making one specific for OctoFarm eventually and I'll be using this one as a base so going forward the databases will work fine to change over when I do make my own. The only thing I'll be changing is getting rid of the database dependency in Filament Manager and linking it with OctoFarms so you only need a single one. 

Thanks as well! Always good to hear :)

NotExpectedYet, 2020-06-17:
Gather this has been answered so closing.

closed 2020-06-17 by NotExpectedYet

tonysurma, 2020-06-18:
Yep thanks for the info