OctoFarm/OctoFarm #71: cant delete or upload files when a print is runinng

Avatarsia, 2020-06-13:
like the topic said.

NotExpectedYet, 2020-06-14:
If this is on Vivaldi browser I'd probably say it was related to your other issue?

NotExpectedYet, 2020-06-14:
I'm also going to need more information. Where can't you upload files? File Manager doesn't show active printers but the other views shouldn't stop uploads.

Avatarsia, 2020-06-14:
it seems its an issue of the upload loop i posted.
because its the same file name that i tried to delte after multiple uploads.
so far i can still see running printers in the file manager ?

NotExpectedYet, 2020-06-14:
Hmmm, I've just checked yeah you can see running printers in file manager. I was thinking to the multiple-upload section sorry. 

Yeah I'm going to close this as I believe it's down to the original upload you did getting messed up in some way. Can discuss further in that thread I think this is just an offshoot of that issue as you mentioned it was the same file.

closed 2020-06-14 by NotExpectedYet