OctoFarm/OctoFarm #72: upload loop

Avatarsia, 2020-06-13:
when i upload a gcode by drag n drop with the vivaldi browser into a folder created by octofarm i get the request  "would you like to print upon upload?", anyway what i press it restart the next upload, when i click yes it prepares the start and try to upload but fails - see issue cant delete or upload anythink when printer is running.

NotExpectedYet, 2020-06-14:
Current supported browsers are Chrome and Firefox. I'd use one of those until I can look into this.

Avatarsia, 2020-06-14:
Version 

Vivaldi | 3.0.1874.33 (Stable channel) (64-Bit)

I think the core is like chrome, no ?

NotExpectedYet, 2020-06-14:
No idea, 

I'll double check and make sure there's nothing wrong with that specific function. Could you try and Chrome or Chromium and see if you get the same results?

NotExpectedYet, 2020-06-14:
I'd Re-Sync as well to clear out any possible issues with the file list getting out of sync.

NotExpectedYet, 2020-08-17:
I'm going to close this currently as Vivaldi isn't currently on the list of supported browsers. Because I'm a one man team I will be only focusing on Chrome and Firefox going forward.

closed 2020-08-17 by NotExpectedYet