OctoFarm/OctoFarm #73: Filament Manager Issues/Experiences

tonysurma, 2020-06-16:
[currently in draft, will clean up in a bit but didn't want to lose my notes]

I will try to add in screenshots in a bit:

1) When I created a new profile the green success message said "successfully added new roll to database" not new profile

2) after that I could not save (no error) a new roll if I choose that profile without first refreshing page (verified that if I picked a previous defined profile I was good to go)  I was able to later edit the roll to choose the profile

3) I have rolls defined/attached to the printers in filament manager in each of the octoprint instances however octofarm doesn't seem to know about them and wants me to pick them separate there...  also not sure yet - will try after a print is done - if I pick via octofarm if it gets assigned in filament manager in octoprint

4) I have a MK3S MMU2S as one of the printers and in filament manager in octoprint I have different rolls assigned to each of the 'tools' as the ui calls them but again they don't show in octofarm and also is asking me to pick one from the octofarm UI - I will try later to see what doing via the UI does

again in draft but wanted to push here for now

NotExpectedYet, 2020-06-16:
1. Not sure about that will take a look. 
2. Be related to the first. 
3. Yeah OctoFarm doesn't currently detect which printers have which in Filament manager but once they are assigned they need to be tracked in OctoFarm. 
4. Nothing as of yet, but the next release will sort this for you. :)

tonysurma, 2020-06-16:
Thanks for 3 - I played a bit - if you assign a spool to a printer in octofarm it sets tool0 to that filament in octoprint/filament manager so that is good/works.

If you choose no selection in octofarm it changes nothing in octoprint/filament manager which is okay as well (i.e. doesn't remove it from the other printer)

As you said octofarm doesn't know about any changes/assignments made in octoprint/filament manager currently.

and lastly for MMU it as I noted in the start of this update - it just always updates tool 0.

So for today for non MMU printer - I can just do selection in OctoFarm and it puts that in octoprint which is great.

For MMU I will do via octoprint and it will do all the tracking just not 'know' about that in OctoFarm which is okay.

NotExpectedYet, 2020-06-17:
Yes that is correct, currently OctoFarm only tracks the tool0 but the current development version changes this. 

Yeah I'm not really sure how to deselect a spool from filament manager as of yet, but that seemed to work fine for my use case. If it become annoying for people not deselecting a spool I'll try and do something to allow for this. 

Regarding your MMU, the next update will allow for selections of all the tools you have assigned in the OctoPrint profile. So as long as you've got multiple nozzle and a single extruder in there you should be good choosing the filament for those. History has been collecting these statistics for a while now so if you want to you'll be able to go assign a used spool (if still available) to octofarms history and back fill it all. Not great, but it works.

NotExpectedYet, 2020-07-19:
Looking for testers, these fixes should be applied. 

Currently only one left is checking what's attached on octoprint

commit 9fd34a6e (2020-07-16): Fixed #73
closed 2020-08-16 by NotExpectedYet