OctoFarm/OctoFarm #74: Filament Manager

lw4dyu, 2020-06-17:
Hello, I installed OctoFarm, and I really liked the dashboard change, and the new features.
I have the filament rolls registered in the "Filament Mannager" and assigned to each printer. The statistic of how many kg you have in the warehouse is very good.
But when printing, I do not discount the use of the filament.
It's okay ?? or can it be fixed ??
Thank you very much for your development.

NotExpectedYet, 2020-06-17:
Hello, thanks for the kind words. 

It currently only discounts the amount after a print is completed as far as I can tell there's no live tracking and seemed a bit much to add in that level of tracking. Is that what you mean? 
Or do you mean it's not down dating after a print has completed? 

If it's the former I'm not sure if it's worth fixing, but your welcome to put a feature request in at features.octofarm.net for one.

lw4dyu, 2020-06-17:
Hello, thank you for your quick response.
She does not discount it, after finishing the impression.

live tracking, I think it's not worth it

NotExpectedYet, 2020-06-17:
Thank you for the clarification I will take a look into it :) 

It doesn't currently re-sync the entire spool database just the used spools on a print so I gather it's not down dating the totals. I'm re-working a lot of that and should have a fix/improvement in the next few days.

lw4dyu, 2020-06-17:
Thank you very much for your project.
I think it is a great project.
I only program in a RAD called Scriptcase. but if I could collaborate I would do it with pleasure.
What programming language do you develop it in?
The only thing I could do is collaborate on Patreon. Later I will do it again.
Thank you

NotExpectedYet, 2020-06-19:
I only currently program this in nodejs so JavaScript. 

Patreon is always fine :) 

I'm using this project to learn full stack application design more so I may be annoying with questions if you did help :P. 

Thanks for the support anyway! I'll make sure to update here when I've squashed that bug :)

lw4dyu, 2020-06-24:
Hello, pass all instances of octoprint and Filament Manager to use postgresql.
synchronized OctoFarm, and began to synchronize the use of the filament.
Thank you very much for the great work. and Apologies for the above "claim".

NotExpectedYet, 2020-07-16:
Sorted this now finally, will be pushed to master as soon as I've double checked all the dual nozzle changes

NotExpectedYet, 2020-07-19:
Looking for testers for this

lw4dyu, 2020-07-19:
I offer to try it

Claudio Alberto CAPRA
movil : +5492216312539
email : ccapra@gmail.com

Follow us:

<https://www.facebook.com/10212826923483425> <https://twitter.com/lw4dyu>
<https://plus.google.com/u/0/+ClaudioAlbertoCapra>
<https://ar.pinterest.com/claudioalbertoc/>
<https://www.linkedin.com/in/claudioalbertocapra/>
<https://www.youtube.com/channel/UCM2ep4Ee6Vc9WWmhfgJxSOA>

NotExpectedYet, 2020-07-19:
> I offer to try it Claudio Alberto CAPRA movil : +5492216312539 email : ccapra@gmail.com Follow us: <https://www.facebook.com/10212826923483425> <https://twitter.com/lw4dyu> [...]

It's on development branch currently :)

lw4dyu, 2020-07-19:
ok, can you update only this module?

Claudio Alberto CAPRA
movil : +5492216312539
email : ccapra@gmail.com

Follow us:

<https://www.facebook.com/10212826923483425> <https://twitter.com/lw4dyu>
<https://plus.google.com/u/0/+ClaudioAlbertoCapra>
<https://ar.pinterest.com/claudioalbertoc/>
<https://www.linkedin.com/in/claudioalbertocapra/>
<https://www.youtube.com/channel/UCM2ep4Ee6Vc9WWmhfgJxSOA>

NotExpectedYet, 2020-07-22:
It's currently only on development yeah, you will have to switch to that branch and try it. I also recommend backing up your database before hand or using a new one.

NotExpectedYet, 2020-08-10:
This should all be resolved on the latest RC

commit 33252379 (2020-07-16): Fixed #77
Fixed #74
Linked back up filament manager to add/remove rolls
closed 2020-08-16 by NotExpectedYet