OctoFarm/OctoFarm #75: Fullscreen on Panel View and Camera View

juangonzalezpr, 2020-06-17:
Hi,
I hope you are well. I found that in Panel View and Camera View at the time of double-clicking the camera box for the fullscreen it changes to fullscreen for a few seconds and then returns to the normal state. It does not stay on fullscreen.

NotExpectedYet, 2020-06-17:
Hey thanks for the report. 

Yeah that's due to the new sorting stuff and an oversight on my part. I've already pushed a fix to development and should be on master within a few days I'm just checking through the multi-tool support I've just added. :)

juangonzalezpr, 2020-06-17:
Thanks for your quick response. I like the interface I am working with 5 machines so far and works like a charm. Can't wait to have more printers.

NotExpectedYet, 2020-06-17:
No worries glad you like it :) 

More cool stuff coming in the pipe line certainly! 

Yeah I feel you, I'm currently on 3 and still feel that bug ahaa

juangonzalezpr, 2020-06-17:
That bug LOL. I know. 

[image]

juangonzalezpr, 2020-06-17:
Any testing feel free to ask. ;-)

NotExpectedYet, 2020-06-17:
Dude that's ace! Looks well organised unlike my own :P I've got 3 different types of printers and there's just something so clean about them all been the same type in enclosures! 

My next bout is with multi-nozzle support so if you've got any with multiple extruders or tools then your welcome to pull development. 

I've got a testing group in our discord and I'm always happy for more input :)

juangonzalezpr, 2020-06-17:
Done! thanks for the invitation.

NotExpectedYet, 2020-06-17:
No worries :) 

I've added you to the development group. 

I ask anyone reporting bugs for a development version to do them in that group and keep here for master versions. Just because half the time the development versions there's something I've forgotten or messed up aaha

juangonzalezpr, 2020-06-17:
hahaha Thanks for that. I keep informing you in the group about anything I find.

NotExpectedYet, 2020-08-10:
This should be fixed on the latest RV

commit 72775f18 (2020-06-17): fix for volume showing too many leading 0
Fixed #75
closed 2020-08-16 by NotExpectedYet