OctoFarm/OctoFarm #76: thumbnail support, where do they show?

tonysurma, 2020-06-18:
I just saw in release notes that it supports the thumbnails and I have been using that pretty extensively for weeks now in octoprint.

However I don't see them show up anywhere in octofarm.  Where/how should they show?  I can then debug but not sure where they should be showing.

Thanks again

NotExpectedYet, 2020-06-18:
Are you on version 1.1.5.5? as they should just show in the file manager. 

[image]

NotExpectedYet, 2020-06-18:
Which thumbnail plugin are you using as well? Currently it's only UFP and Prusa.

tonysurma, 2020-06-18:
looks like just 1.1.5 not 1.1.5.5, I need to set a reminder to check for new versions more often (or maybe we can get a tickler in the app?

And using Prusa.  Updating now and my apologies for not noticing version

tonysurma, 2020-06-18:
closing this as it totally works if you are on the right version :)

closed 2020-06-18 by tonysurma