OctoFarm/OctoFarm #77: Farm Utilisation - Offline Hours Calculation incorrect/unclear

Schokobecher, 2020-06-21:
The **Offline Hours** within the **Farm Utilisation** are inaccurate or what they are based on is not clear to me. They keep ticking up irregularly (every 5 to 10+ seconds) even though all connected printers are either Active or Idle - **Current Status** and **Current Utilisation** statistics seem correct and don't have this issue.

A fresh install with printers added looks like this after about 2 minutes
[image]

While  **Current Utilisation** looks like this:
[image]

An install left running for about 7hrs looks like this with two printers connected and online during that time. I would expect the ~65% added to **Idle Hours**, not **Offline Hours**

[image]
[image]

NotExpectedYet, 2020-06-21:
Cheers for that, I'll work in some improvements for the next release

Schokobecher, 2020-06-22:
Not sure if related, but **Current Utilisation** also has gone wacky over time:

[image]
[image]

So 50% active + 1% Offline = 49% of missing (most likely idle) in the pie chart

NotExpectedYet, 2020-06-28:
> Not sure if related, but **Current Utilisation** also has gone wacky over time:
> 
> [image]
> [image]
> 
> So 50% active + 1% Offline = 49% of missing (most likely idle) in the pie chart

Yeah I've just found the reason for that and I'm confused to why the issue wouldn't present itself sooner. I'm just doing some tweaks to sort all this for the next version. Will collect actual values now rather than trying to work it out taking active time from time alive.

NotExpectedYet, 2020-07-19:
This should be resolved, looking for people to test the new RC

commit 6c08670c (2020-06-29): added in current operations and statistics
Fixes #77
commit 33252379 (2020-07-16): Fixed #77
Fixed #74
Linked back up filament manager to add/remove rolls
closed 2020-08-16 by NotExpectedYet