OctoFarm/OctoFarm #79: Duplicate Entries in history

Simmons928, 2020-06-30:
I have noticed that I will get duplicate entries in the history
[image]

It appears, in the interface at least, that they are separate items. 
[image]

Deleting one of the duplicates will not affect the other. 

if there is anything I can collect or provide to assist, let me know.

NotExpectedYet, 2020-07-01:
The duplicates are caused by multiple listeners, so the websockets don't seem to be closing properly.

Have these printer been Online -> Offline or Edited at any point? I was getting it at 1 point due to the ping/pong check but I am pretty sure I resolved that in the latest version. 

Your state server log would be good to look through if you don't mind. You can download it from "System" -> "Server Settings" -> "Logs"

A server restart should clear the listeners, and I'll try and get a fix ready for next release.

Simmons928, 2020-07-01:
See below. 
I believe I see what you are referring to in the log. 

`Sun, 28 Jun 2020 20:16:53 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.201:5000/sockjs/websocket | 
Sun, 28 Jun 2020 20:16:53 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.201:5000/sockjs/websocket | 
Sun, 28 Jun 2020 20:16:53 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.201:5000/sockjs/websocket | 
Sun, 28 Jun 2020 23:43:01 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.201:5000/sockjs/websocket | 
Sun, 28 Jun 2020 23:43:01 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.201:5000/sockjs/websocket | 
Sun, 28 Jun 2020 23:43:01 GMT | INFO | OctoFarm-State.log | PrintDone0: ws://192.168.2.201:5000/sockjs/websocket | `

[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/4857825/OctoFarm-State.log)

NotExpectedYet, 2020-07-01:
Yeah after editing printers. 

Cool, I think I've squashed it on development so I'll do some more testing to double make sure.

Simmons928, 2020-07-01:
Awesome! 
As a workaround, I will recycle the docker for OctoFarm in the meantime after I edit any of the printers.

NotExpectedYet, 2020-08-10:
This should be fixed on the latest RC if your available to test

Simmons928, 2020-08-10:
I will test as soon as able.

Simmons928, 2020-08-11:
I am running the OctoFarm-Monolithic docker and it does not appear to be updated at this time. (At least in the repo I am using octofarm/octofarm:monolithic-latest) I am willing to update to test but I am not sure if there is another repo for this.

NotExpectedYet, 2020-08-11:
There's a tag called :dev you can replace :latest with on your UnRaid docker settings. Will pull the latest. 

There's still a damn double listener been created and I'm not sure wear at the moment. It isn't "edit" or "delete" though.

Simmons928, 2020-08-11:
I was able to update to the dev release. 
I was not able to test as I ran into an issue with the websockets.
It seems that this release does not play nice with OctoPrint 1.4.2 
I went back to 1.1.5 and it began working again and it was connecting with the clients. 

[image]
[image]

Printers get stuck in state "Waiting for Websocket" 

`Tue, 11 Aug 2020 20:22:13 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 0: ws://aci3ms-01.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:14 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 502 | data:"0: ws://aci3ms-01.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:14 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 0: ws://aci3ms-01.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:14 GMT | ERROR | OctoFarm-State.log | WebSocketClient: Error | data:{"0":{}} | 
Tue, 11 Aug 2020 20:22:37 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"1009: ws://ce3-03.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:37 GMT | INFO | OctoFarm-State.log | WebSocketClient: Closed | data:{"0":1006} | 
Tue, 11 Aug 2020 20:22:42 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 3: ws://ce3-03.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:42 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 502 | data:"3: ws://ce3-03.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:42 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 3: ws://ce3-03.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:42 GMT | ERROR | OctoFarm-State.log | WebSocketClient: Error | data:{"0":{}} | 
Tue, 11 Aug 2020 20:22:44 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"1007: ws://ce3-01.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:44 GMT | INFO | OctoFarm-State.log | WebSocketClient: Closed | data:{"0":1006} | 
Tue, 11 Aug 2020 20:22:47 GMT | INFO | OctoFarm-State.log | WebSocketClient: retry in 5000ms | data:"1008: ws://ce3-02.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:47 GMT | INFO | OctoFarm-State.log | WebSocketClient: Closed | data:{"0":1006} | 
Tue, 11 Aug 2020 20:22:49 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 1: ws://ce3-01.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:50 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 502 | data:"1: ws://ce3-01.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:50 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 1: ws://ce3-01.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:50 GMT | ERROR | OctoFarm-State.log | WebSocketClient: Error | data:{"0":{}} | 
Tue, 11 Aug 2020 20:22:52 GMT | INFO | OctoFarm-State.log | Re-Opening Websocket: 2: ws://ce3-02.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:52 GMT | ERROR | OctoFarm-State.log | Error: Unexpected server response: 502 | data:"2: ws://ce3-02.xxxxxx.xxx/sockjs/websocket" | 
Tue, 11 Aug 2020 20:22:52 GMT | ERROR | OctoFarm-State.log | WebSocket hard failure: 2: ws://ce3-02.xxxxxx.xxx/sockjs/websocket | 
Tue, 11 Aug 2020 20:22:52 GMT | ERROR | OctoFarm-State.log | WebSocketClient: Error | data:{"0":{}} | 
`

[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/5059330/OctoFarm-State.log)

1.1.5 worked with out any issues. 

Let me know if there is anything else you need from me.

NotExpectedYet, 2020-08-12:
Thanks, for 1.4.1+ you'll need to use the User API Key or the Application API key just for completeness, sorry I didn't expect farm owners to update OctoPrint so fast. 

1.1.5 OctoFarm is working with 1.4.2? that shouldn't be the case from my understanding of the issue as I just explained ahaha. Or did you mean when you downgraded? 

The files is something been worked on currently. The old file system would time out on larger file systems so I've told the system not to wait for the response and just continue without it. There's 1 more check that needs adding to make that totally obvious to the user and not hidden in the new printer settings modal. Thanks for the detailed report too!

Simmons928, 2020-08-12:
My comment above got a little confusing because I made it in the middle of trying to identify exactly what problems I was actually running into. 

To clarify my versioning. 

I was running 1.4.0 on all my OctoPrints with 1.1.5. 
I had just added the Creator Pro to the farm and it was running 1.4.1 and it seemed to work in 1.1.5 (Though it was popping in and out randomly)
I updated all of my machines to 1.4.2 thinking that might solve my problem and I had been putting off the updates precisely because everything was working... the update to 1.4.2 broke everything and that is when I found the outstanding issues #88 and #85.

1.1.5 was working with the 1.4.0 machines but not 1.4.1+
I am now running 1.1.5.6.8 with all my OctoPrints rolled back to 1.4.0. 
The file system seems to be working, though the initial start seems to take a bit of time for the files to appear in OctoFarm.

NotExpectedYet, 2020-08-12:
> My comment above got a little confusing because I made it in the middle of trying to identify exactly what problems I was actually running into.
>
> To clarify my versioning.
>
> I was running 1.4.0 on [...]

Yeah they go through some extra processing now to speed up the client, but future server restarts should be faster as the files are now stored in the database. Thanks for the clarification. 

1.1.5 might possibly work if you use an API key from the User section or a specific Application key. That's the current issue as I mentioned above, I've just not had chance to test the theory. Going forward I'm going to be recommending people use that anyway, and the UI will warn a user albeit that just needs to be made a little more obvious. I'm also going to put a (currently processing...) box in the file manager which should help the initial confusion with booting and not seeing files.

commit a749e7e8 (2020-07-01): updated to keep jobs in sync with new formatting
Fixed #79
closed 2020-08-16 by NotExpectedYet