OctoFarm/OctoFarm #81: Sort Index isn't working

nuubone, 2020-07-02:
Im having a problem to organize my printers, I've tried to organize them by the sort index, but it doesn´t work, it always back to default. 
How I can solve this?
I would like to recommend a filter by printer name, it would be practical.
Ty for all

NotExpectedYet, 2020-07-02:
Could you expand on that a little more please? Is there an issue with you updating the sort index or is it not applying the sort index? 

The sort index is the default and decides the initial order of the printers on the views. The views each have an override to be applied and that should always revert back to the sort index. Do you use any of the filters at the top of views?

nuubone, 2020-07-02:
I change the order and the sort index change. But when I reload the page, the index number keep in that printer, but it is not in sequence.
To keep this sequence, I have to restart the server.
Idk how to upload the image here, so I do that on imgur
[https://imgur.com/3pwl6tq](url)

NotExpectedYet, 2020-07-02:
Thanks for the reply 

Oh, so basically that should have already reflowed and re-ordered on the page, somethings not right especially if your having to restart the server. I'll make sure I can't reproduce this on dev and let you know. I think it's something I'll have to fix btw so can't really make any recommendations unless your versed in editing a mongo database.

Oh forgot to mention as well, I've no formal plans for filter by printer name on views as of yet. If you'd like to log a feature request here I'll certainly look into it. 
https://features.octofarm.net

NotExpectedYet, 2020-08-10:
Hello, I've fixed some bugs around this issue with the latest RC, would you be available to test it?

nuubone, 2020-08-11:
oh, Sure. I will check it soon.
This new version is already stable?

NotExpectedYet, 2020-08-12:
> oh, Sure. I will check it soon.
> This new version is already stable?

It's getting there, there's a bug in retraction and some other things like been allowed to delete attached filament but mainly it's fine.

NotExpectedYet, 2020-08-16:
Master has been updated could you make sure this is fixed please?

NotExpectedYet, 2020-09-19:
This has been stale for some time now, I'm closing. Please re-open if the issue is still persistent.

closed 2020-09-19 by NotExpectedYet