OctoFarm/OctoFarm #83: Filament Page became corrupt

ruderebel, 2020-07-20:
I was adding some new filament which  went fine, but then I tried to edit on of the filament brands which I incorrectly spelled and now the Filament page is dead.  When I try to access it, I get the following:
[image]

I now cannot access this page and manage my filaments.

NotExpectedYet, 2020-07-22:
Is this with filament manager enabled or disabled? 

Have you tried re-starting the server since the issue occured?

NotExpectedYet, 2020-07-22:
Also what's the version number please

ruderebel, 2020-07-23:
Sorry for the delay.  

I am currently on 1.1.5.5, but am just about to jump to the next version for testing for you, but can hold off if you need.

Yes, I have retried restarting the server (2 different times) and nothing.  The image I posted is literally what my Filament tab looks like when I click the link now.

NotExpectedYet, 2020-07-23:
> Sorry for the delay.
>
> I am currently on 1.1.5.5, but am just about to jump to the next version for testing for you, but can hold off if you need.
>
> Yes, I have retried restarting the server (2 [...]

You can give it a try certainly. 

I'm hoping the action didn't corrupt the database. So was it with the filament manager plugin disabled? I gather not it's just two different parts of the code is all.

ruderebel, 2020-07-26:
ok, so I have an update. 

I upgraded to 1.1.5.6.7 (development branch) for testing and it seems that the Filament manager is back up and running.  Here is the log I downloaded (unfortunately, I forgot to download right after starting up, so this log would be after I did some further changes in the tab.
[OctoFarm-FilamentManager.log](https://github.com/NotExpectedYet/OctoFarm/files/4977015/OctoFarm-FilamentManager.log)

NotExpectedYet, 2020-07-26:
Thanks for testing that!

That looks fine to me apart from it couldn't grab the spools from your printer when you turned on filament manager sync? 

That will remove all your pre-entered filament just to warn.

NotExpectedYet, 2020-07-26:
Proper jealous you've got some PC btw

ruderebel, 2020-07-26:
So, when I initially entered, the filament brands I tried to enter were gone (no big deal as there were only 2 deleted).  I added a bunch of new stuff, I can't remember why, but I needed to reboot my PC, and when I reentered OctoFarm, all my filaments were gone, including all of the newly entered ones.  No biggie, I reentered everything and now it is golden.

yeah, I made a decent filament order last week and bought a few "pricier" filaments to try out some things.  Getting ready to dabble in PC and slowly upgrading my AnyCubic Predator to be able to print it.  That printer is a delta and the frame is such that I was able to build a removable enclosure using nothing but the printer's frame, foam board, and a couple of printed pieces.  Just need to find a good way to control the temp inside.  The PC I got is a blend which allows you to print on the lower temps than the usual PC made by a local maker and it didn't break the bank :)  My costs are in CAD, by the way, in case you were wondering so those are actually decent prices.

closed 2020-07-26 by ruderebel