OctoFarm/OctoFarm #84: No History yet - Sync History with

quloquloqulo, 2020-07-22:
I don't know why but my hystory is always empty reporting "No History yet Come back here when you've completed some prints."...

it would be interesting to remedy the problem if it could be synchronized with the printjobhistory plug-in and also recover old prints that octofarm has not saved

NotExpectedYet, 2020-07-22:
Can you pass me some history logs please?

NotExpectedYet, 2020-07-22:
And also what's the version number please?

quloquloqulo, 2020-07-22:
[OctoFarm-HistoryCollection.log](https://github.com/NotExpectedYet/OctoFarm/files/4960052/OctoFarm-HistoryCollection.log)

NotExpectedYet, 2020-07-22:
> [OctoFarm-HistoryCollection.log](https://github.com/NotExpectedYet/OctoFarm/files/4960052/OctoFarm-HistoryCollection.log)
> 
> Version 1.1.5.5
Thanks!

Have you setup anything in the filament manager at all? It's looking like this is a bug. 

Regarding syncing history, I've no plans too add support for that plugin but you can make a feature requests at https://features.octofarm.net and I can gauge popularity for it.

quloquloqulo, 2020-07-22:
Thanks for your quick response. 
The filament manager was correctly set on both octoprint and octofarm. However I just disabled and re-enabled sync. We'll see if it resolves with the next print,

NotExpectedYet, 2020-07-22:
Welcome :) 

Hmmm that's just a flag so it shouldn't do much, unless filament was selected when you actioned the disable and re-enable. 

If the spool has selected fine (ie stays there on a page refresh) you should be fine going forward. 

I'm just hoping the database hasn't gotten out of sync. 

History will trigger 10 seconds after a print so if you still don't see if, fire me the logs and I'll double check what's going on.

NotExpectedYet, 2020-08-04:
Hello  did the issue resolve?

quloquloqulo, 2020-08-04:
Yes. I start the prints from the octoprint web interface instead of octofarm. Could this be the problem?

NotExpectedYet, 2020-08-10:
It shouldn't make a difference. 

I think the problem is with the filament manager plugin. I've done a lot of bug fixing around that area for the latest RC. I believe upgrading should solve your problem. It's the last RC before I push the new release currently, I just want to make sure it's continued to be stable as a lot has changed on the backend.

NotExpectedYet, 2020-09-19:
This has been stale for some time now, please re-open if the issue is stiller persisting.

closed 2020-09-20 by NotExpectedYet