OctoFarm/OctoFarm #85: OctoPrint 1.4.1 and Websocket Error

jneilliii, 2020-07-24:
Appears there may be something that is going to break between OctoFarm and OctoPrint 1.4.1. Websocket connections show connected but in a tentative state. Posted in the RC3 feedback ticket, linked below.

[image]

https://github.com/OctoPrint/OctoPrint/issues/3643#issuecomment-663303817

jneilliii, 2020-07-24:
[OctoFarm-Server.log](https://github.com/NotExpectedYet/OctoFarm/files/4970139/OctoFarm-Server.log)

jneilliii, 2020-07-24:
[OctoFarm-State.log](https://github.com/NotExpectedYet/OctoFarm/files/4970141/OctoFarm-State.log)

NotExpectedYet, 2020-07-24:
Yeah nothing in your logs, it send that auth fine but never setup the listeners/got any data.

NotExpectedYet, 2020-07-24:
From my end it seems the websocket is connecting but looking at OctoPrint logs I received the following: 

```
2020-07-24 06:39:52,627 - octoprint.server.util.flask - INFO - Passively logging in user s1mpleman from 192.168.1.50
2020-07-24 06:39:52,627 - octoprint.access.users - INFO - Logged in user: s1mpleman
2020-07-24 06:39:52,654 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.50
2020-07-24 06:39:53,098 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.50
2020-07-24 06:39:53,118 - octoprint.server.util.flask - INFO - Passively logging in user s1mpleman from 192.168.1.50
2020-07-24 06:39:53,119 - octoprint.access.users - INFO - Logged in user: s1mpleman
2020-07-24 06:39:53,653 - octoprint.server.util.sockjs - INFO - User s1mpleman logged in on the socket from client 192.168.1.50
2020-07-24 06:39:57,795 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.1.50
2020-07-24 06:39:57,878 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.50
2020-07-24 06:39:57,881 - octoprint.server.util.sockjs - WARNING - Unknown user/session combo: s1mpleman:{removedAPIKEY}
```

The API checks all pass fine with the same key. I've tried with global keys/application keys and user specific keys all gave me the unknown user/session combo. Hope that helps.

foosel, 2020-07-24:
So... I'm pretty sure that is actually a bug in OctoFarm. It shouldn't be using the **API key** but a **session key**. That using an API key even worked before was a bug in OctoPrint. What you should be doing is get a session key for the API key via a passive login with the API key, and then use the session from that. It also says so in [the docs](https://docs.octoprint.org/en/master/api/push.html#push-updates):

[image]

An example for that can be found in the recovery dialog recently added to the `maintenance` branch:

https://github.com/OctoPrint/OctoPrint/blob/87cd0a88fbd370f3ef760501f9518663879adba8/src/octoprint/static/js/recovery/recovery.js#L56-L63

NotExpectedYet, 2020-07-24:
> So... I'm pretty sure that is actually a bug in OctoFarm. It shouldn't be using the **API key** but a **session key**. That using an API key even worked before was a bug in OctoPrint. What you [...]

Makes sense thanks I'll take a look. 

Aye I saw the message and just presumed the session key was an API Key, don't have a great understanding of them tbh. I'll update OctoFarm to work with the session key instead.

commit dce458eb (2020-08-08): fixed connection issues with 1.4.2
Fixed #85
Fixed #89
closed 2020-08-16 by NotExpectedYet