OctoFarm/OctoFarm #86: Errors with NodeJS 14.6.0

jariivanoff, 2020-07-28:
No webpage, ERR_CONNECTION_TIMED_OUT.
Content of PM2 app-error.log:

(node:18820) Warning: Accessing non-existent property 'count' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
(node:18820) Warning: Accessing non-existent property 'findOne' of module exports inside circular dependency
(node:18820) Warning: Accessing non-existent property 'remove' of module exports inside circular dependency
(node:18820) Warning: Accessing non-existent property 'updateOne' of module exports inside circular dependency
(node:838) Warning: Accessing non-existent property 'count' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
(node:838) Warning: Accessing non-existent property 'findOne' of module exports inside circular dependency
(node:838) Warning: Accessing non-existent property 'remove' of module exports inside circular dependency
(node:838) Warning: Accessing non-existent property 'updateOne' of module exports inside circular dependency

App is running though:
[image]

NotExpectedYet, 2020-07-28:
Legend thanks for logging this. I'll get it checked for the next release.

NotExpectedYet, 2020-09-29:
To keep versions and other things in check I am going to make sure that OctoFarm works with the current LTS versions of nodejs. 

I will close this once all the documentation has been changed to reflect this.

closed 2020-09-29 by NotExpectedYet