OctoFarm/OctoFarm #87: OctoFarm install on MacOs (Mojave, for instance)

bolsoncerrado, 2020-08-05:
I went ahead and tried to install OctoFarm on MacOS Mojave (I later found [this guide](https://gist.github.com/ajatoledo/b576e77fcda95d6d769403cb36f855df)), but seems mongoDB is no longer available as an installable package?

`iMac:~ alex$ brew install mongodb-community
Error: No available formula with the name "mongodb-community"
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
iMac:~ alex$`

Any clues?

Thanks!

NotExpectedYet, 2020-08-08:
I've no experience with Mac systems sadly and that documentation. All I could suggest is making sure brew update is run looking at the documentation. 

Have you checked the official mongodb community website on how to install on Mac?

bolsoncerrado, 2020-08-10:
Autoreplying myself: https://docs.mongodb.com/manual/tutorial/install-mongodb-on-os-x/

closed 2020-08-10 by bolsoncerrado

NotExpectedYet, 2020-08-10:
Thank you for getting back :)